luc-github/Repetier-Firmware-4-Davinci #326: SD Cards gets unmounted after print / after stop

z1cb4, 2018-10-17:
Hey,

i have a Toshiba FlashAir WiFi sd-card in my daVinci.A
After the printer has finished a printjob or after pressing the Stop command from
the printer menu, the files on the card are not accessible anymore from the printer.
I need to shut off the machine in order to able to access the files again.
Any Ideas ?

thnx

z1cb4, 2018-10-18:
I am using 0.92.10
M21 brings the card back!

luc-github, 2018-10-18:
Check you do not have any M22 code in your end code

z1cb4, 2018-10-19:
there is no M22 in my end code script 👍 

G28 X0 ; home the X-axis
G28 Y0;
M104 S0 ; turn off heaters
M140 S0 ; turn off bed
M84 ; disable motors

thank you anyway!

luc-github, 2018-10-19:
I also have toshiba but never got such issue. 
I check next week when back home if I can reproduce. Now I am 10k km from my printer and nothing popup to my eyes when reading
Additional question does this happen when printing from normal micro sd ?

z1cb4, 2018-10-19:
i never tried printing on sd-card before i bought the toshiba card - i will try to reinsert the original sd card that came with the printer! thanks again for your support

z1cb4, 2018-10-28:
sorry was AFK for a few days.
I just tried my old sd card and everything works flawlessly.
There must be an issue with my toshiba flashair or with my configuration.

This is my config file:

[Vendor]

CIPATH=/DCIM/100__TSB/FA000001.JPG
VERSION=F15DBW3BW4.00.02
CID=12344d535731364754d575577b012201
PRODUCT=FlashAir
VENDOR=TOSHIBA
MASTERCODE=123456123456
DNSMODE=1
APPMODE=5
APPSSID=MySSID
APPNETWORKKEY=********
LOCK=1
WEBDAV=2
TIMEZONE=8
UPLOAD=1
NOISE_CANCEL=2

z1cb4, 2018-10-28:
one more thing that i've just noticed:

with the original sd card the printers menu is really snappy whereas after inserting flashair, the printer becomes a bit sluggish.

luc-github, 2018-10-29:
it could be also an issue with your cable adapter it may be too long 

about your config - I do not use webdav with my toshiba not sure if related you can check my repo : https://github.com/luc-github/Flashair-FM/tree/devt

z1cb4, 2018-10-31:
the issue was caused by enabling WEBDAV on the flashair.
i will try to develop a solution using a hotfolder that will automatically
push the files to the flashair.

luc-github, 2018-10-31:
Ok great, so I guess issue can be closed,
Good luck with your project

closed 2018-10-31 by luc-github