luc-github/Repetier-Firmware-4-Davinci #328: Da Vinci 1.0A Motherboard Repair

phimmerich, 2018-11-16:
I have searched high and low but not found this addressed directly. Apologies if it has, please direct me to the link.

I have a Da Vinci 1.0A that will not heat the build plate. The unit was obtained from ebay as non-functioning, but i like to tinker and took the challenge.

Status is that the unit powers on and there are 12V at the connector for the built plate heater. From what i understand, the motherboard/chip switches the ground on and off to cycle the heater on and off. 

I am in the process of tracing back to the ATMEL chip whether there is a component between the chip and the MOSFET is busted or if the chip itself is fried. 
Does anyone have a schematic of the motherboard or pin-out map that lists which line from the chip correspond to which printer component ?

Also are there any spare lines on the chip that we can implement ? 

thanks 

Peter

luc-github, 2018-11-16:
There is no schematics as far I know
Heater bed pin for 1.0A is pin 67 = PB16 which should be PIN 77 on MCU according SAM3X8E spec sheet
Check if you have 3 led on when power on the board - if only 2. it means you have one fuse blowed - to fix it there are several tutorials to repair then on volitivo forum and soliforum

phimmerich, 2018-11-16:
All three LEDs are on.
Are the pins on the MCU normally high and get pulled low when the pin gets activated ?

luc-github, 2018-11-17:
No I think their state is depending of usage, others they may be floating

luc-github, 2018-11-26:
I guess issue can be closed

closed 2018-11-26 by luc-github

phimmerich, 2018-11-26:
Apologies for not following up. Yes the issue can be closed.
Turns out it was operator error with respect to the build plate not heating up. The actual issue with the motherboard was the extruder heater Mosfet. I soldered a replacement part on and everything works now. 
Thanks

luc-github, 2018-11-26:
If you solved your issue - it is all good ^_^