luc-github/Repetier-Firmware-4-Davinci #330: Printing on air

Egauss, 2018-11-24:
;Generated with Cura_SteamEngine 15.01
; Default start code
G28 ; Home extruder
G1 Z15 F100
M107 ; Turn off fan
G90 ; Absolute positioning
M82 ; Extruder in absolute mode
M190 S100
; Activate all used extruder
M104 T0 S230
G92 E0 ; Reset extruder position
; Wait for all used extruders to reach temperature
M109 T0 S230
;Layer count: 249
;LAYER:0
M107
G0 F9000 X95.963 Y74.700 Z0.200
G0 X96.433 Y76.254
;TYPE:SKIRT
G1 F1200 X96.432 Y76.254 E0.00004

after last command printer brings response with Z 15.000 and do the job but on air

;Generated with Cura_SteamEngine 15.01
; Default start code
G28 ; Home extruder
G1 Z15 F100
M107 ; Turn off fan
G90 ; Absolute positioning
M82 ; Extruder in absolute mode
M190 S100
; Activate all used extruder
M104 T0 S230
G92 E0 ; Reset extruder position
; Wait for all used extruders to reach temperature
M109 T0 S230
;Layer count: 249
;LAYER:-2
;RAFT
M126 T0 ; value = 255
;LAYER:-1
;RAFT
;LAYER:-1
;RAFT
;LAYER:0
M127 T0
;LAYER:1
M126 T0 ; value = 40
G1 F2400 E-4.00000
G0 F9000 X92.387 Y40.328 Z0.900
;TYPE:WALL-INNER
G1 F2400 E0.00000

after this bit of code printer prints on the surface as normal

problem with firmware i think, as first bit of example is with first layer 0.2mm
second without supports, brims etc. 
from first code something brings printer response with z 15.000

da vinci 1.0 label but PCB is new shape so google says 1.0A

Egauss, 2018-11-24:
ps same model STL change first layer 0.1mm no supports all ok. same model STL first layer more like 0.1mm  with supports z up 15.000

luc-github, 2018-11-24:
You write print is working depending slicer options ?

Egauss, 2018-11-27:
Sorry for late, printer trying to print with any slice settings, just in the air, now I am just copy paste the the good front end code and its prints on the right level, if I am using just cura generated code it's brings home coordinates and starts print in the air, but not with good front end code, with good code firmware doesn't bring home coordinates and print on right level

luc-github, 2018-11-27:
Slicer need correct settings and start code. default settings always need to be tuned to match printer characteristics.

luc-github, 2018-12-09:
I guess issue can be closed

closed 2018-12-09 by luc-github