luc-github/Repetier-Firmware-4-Davinci #331: Unable to print using extruder 2 with unleaded extruder 1

lcintron, 2018-12-09:
I am unable to print using extruder 2 in a DaVinci 2.0 Duo with extruder 1 unloaded. Tried modifying Cura and Slic3r settings to only utilize extruder 2. Output G-Code uses the right extruder (T2), Ext 2 heats up to temperature while Ext 1 is off yet print won't start until Ext 1 is loaded.  Keep getting a Print Pause - Load Filament on Extruder 1 message that I can't bypass on Repetier Host.

The filament load sensor in extruder 1 (left) seems to be damaged, is there a way to disable it on firmware?

luc-github, 2018-12-09:
T0 is extruder 1 (E0)
T1 is extruder 2 (E1)
you should not have T2 
The filament sensor is trigered if any call to T0
if it is filament sensor issue - you can disable it in Printer menu settings

luc-github, 2019-01-19:
I guess Issue can be closed as no update

closed 2019-01-19 by luc-github