luc-github/Repetier-Firmware-4-Davinci #332: inactive time

tecan, 2018-12-19:
if the printer runs out of filament it should cool down using inactive time in eeprom. min stayed on the filament load screen at full temperature for over 10 min.

luc-github, 2018-12-19:
This is good idea for extruder - what about the bed ?

tecan, 2018-12-19:
for the hotbed maybe drop the temp to 40 deg C and beep periodically so PLA parts dont fall off or just disable it and beep. if you wanted to set that longer maybe 30 min for the hotbed and 5 min for the hotend ?

luc-github, 2018-12-19:
well this will not work for ABS a little temperature change and ABS will not like

luc-github, 2018-12-19:
No heat bed for ABS, really ? I use 110 degres even with 3M glue

luc-github, 2019-03-18:
@tecan Did you pushed this to repetier main FW

tecan, 2019-03-18:
i dont typically follow what the main repo does, is that were alot of the new changes come from still ?

luc-github, 2019-03-30:
Yes I try to stick to repetier main FW as easier to maintain - unless it is Davinci specific

tecan, 2019-04-23:
looking in void Printer::defaultLoopActions()   i see some isMenuMode(MENU_MODE_SD_PAUSED)  maybe all we have to do is add in the filament change menu

luc-github, 2019-04-24:
you mean to count pause time as idle time ?

luc-github, 2019-04-24:
I need to read the code again but - I am pretty sure that when just heating and waiting, like for preheat and wait temperature stable  - and not printing the inactivity counter is not take in account.
same if you are in pause and you cut temperature for any reason so your extruders are cold, then you also heat again and wait temperature is stable, if for example inactivity time is 1 min and we still are heating what will happen ?
I think implementation is not that simple - need to take in account all user cases - I am pretty sure also the flag MENU_MODE_PRINTING is enabled when heating so check  only MENU_MODE_SD_PAUSED is not enough.
Also need to allow user to not use this feature is also important

I am not against - it is good idea but not that simple IMHO if you consider all possible user cases

tecan, 2019-07-09:
is this still fire hazard ?

luc-github, 2019-07-09:
Just a bot that automatically check issues

tecan, 2019-08-20:
bump.... this is annoying and very important

luc-github, 2019-08-20:
what is repetier answer ? may need a bump there too ^_^

tecan, 2019-09-11:
i might have a look at this issue next month, no response from repetier yet.

luc-github, 2019-09-11:
Yes may be because issue from custom FW and not pure repetier
I had a quick look and I think need to setup specific trigger for this - I add a flag to keep issue open as I have have really no time now

netpipe, 2021-07-11:
hmmm, winter is coming. might get to it then, it is a fire hazard so i'de mark this as critical.