luc-github/Repetier-Firmware-4-Davinci #333: DAVINCI 1.0 using ZMIN Evaluation results in nozzle striking bed side when printing

sbeaman, 2018-12-26:
After performing "MAINTENANCE/Zmin evaluation" the resulting value causes the print nozzle to strike the side of the bed when trying to print. (When leaving home X,Y)

The issue appears to be Z_PROBE_HEIGHT is set to 0.28 and should be 0. (the difference between activated probe position and nozzle height)
Reason:
The z probe is offset from the nozzle by -0.28. The probe touches a metal plate that is 0.28 above the bed. This means when the probe touches the calibration plate the nozzle is at bed height (0 offset).

The current value of 0.28 causes the bed to travel vertically too high. (The probe is at bed height). When the head leaves home the probe clears the bed BUT the nozzle strikes the side of the bed and is unable to move.

P.S.
Thanks for the awesome work of Repetier support for the DaVinci. It is greatly appreciated.

luc-github, 2018-12-26:
@sbeaman what version of FW do you use ?
Can you confirm your Davinci board : 1.0 or 1.0A ?
Can you describe what action you do to make  
>the nozzle strikes the side of the bed and is unable to move.

I need clear steps / info to reproduce the issue

sbeaman, 2018-12-26:
Thank You for your response.

It is the Davinci 1.0 board. (Short JP1 to reset EEPROM)
Repetier- Host V2.1.3
Slice with CuraEngine
Steps:
1) Select the menu option "Maintenance/Zmin evaluation"
2) Select "NO" for question "Do you want to clean nozzle and plates first"
3) Allow the printer to measure the three calibration points
4) Press "OK" to move to the next step
5) Select "Yes" for question "Save to EEPROM"
6) Select "No" for question "Do it again?"
7) Launch Repetier-Host and connect to the printer
7) Send G-Code G28 to home the nozzle
8) Print any object. When the head leaves home position:
    - The z-probe will clear the bed
    - The nozzle will strike the side of the bed
    - The head won't be able to move
    - The printer will make horrible noises

Assumptions:
  - Bed is level
  - No G-Codes are issues that impact Z position or Zmin

luc-github, 2018-12-26:
Thank you,
I still need the FW version you are using on printer
I also need the start gcode of your slicer

sbeaman, 2018-12-28:
No Problem,
FW from GIT-HUB: Davinci-Beta_Mar_15_2018_v0.92.10rc4

G-Code: (It is the default for CuraEngine. I added the G28 at start)
`; Default start code
G28 ; Home extruder
G1 Z15 F{Z_TRAVEL_SPEED}
M107 ; Turn off fan
G90 ; Absolute positioning
M82 ; Extruder in absolute mode
{IF_BED}M190 S{BED}
; Activate all used extruder
{IF_EXT0}M104 T0 S{TEMP0}
G92 E0 ; Reset extruder position
; Wait for all used extruders to reach temperature
{IF_EXT0}M109 T0 S{TEMP0}`

luc-github, 2018-12-28:
Thank you,
 so looking at start code, your bed is going down 15mm after G28, so bed should not block your noozle in that case:
>The nozzle will strike the side of the bed

I try to understand when the nozzle will hit bed as you describe 
can you clarify ?

sbeaman, 2018-12-28:
Yes.
- The bed lowers 15mm while the nozzle and bed warm up.
- Once the temperature is reached, the bed moves to printing height.
- The head leaves home X,Y traveling to start printing (normally center of the bed)
- The Z-Probe will clear the bed
- The nozzle strikes the side of the bed. Hitting the glass top.
- The head is unable to move

luc-github, 2018-12-28:
Sorry still not clear
>The head leaves home X,Y traveling to start printing (normally center of the bed)
Ok we are middle of the bed the print should start
>The Z-Probe will clear the bed
>The nozzle strikes the side of the bed. Hitting the glass top.
What command do this ? should be part of your start code but I do not see it

sbeaman, 2019-01-01:
Woops. The table move I added to try and stop the issue. You just need to issue a G28 to home. The issue occurs when you start printing at zmin. Below is the start code without the 15mm bed move:

; Default start code
G28 ; Home extruder
M107 ; Turn off fan
G90 ; Absolute positioning
M82 ; Extruder in absolute mode
{IF_BED}M190 S{BED}
; Activate all used extruder
{IF_EXT0}M104 T0 S{TEMP0}
G92 E0 ; Reset extruder position
; Wait for all used extruders to reach temperature
{IF_EXT0}M109 T0 S{TEMP0}

When the print head moves the image below shows what happens:
https://drive.google.com/file/d/1WeqQqY0UhzlpYkdpPPkKsiMyh_DTcyfr/view?usp=sharing
The Z-Probe clears but the head strikes the glass.

By Adjusting the Z-min by 0.28  the head passes:
; Default start code
G28 ; Home extruder
G1 Z0.28; Correct Z by Z_PROBE_HEIGHT
G92 Z0 ; Set new Z home
M107 ; Turn off fan
G90 ; Absolute positioning
M82 ; Extruder in absolute mode
{IF_BED}M190 S{BED}
; Activate all used extruder
{IF_EXT0}M104 T0 S{TEMP0}
G92 E0 ; Reset extruder position
; Wait for all used extruders to reach temperature
{IF_EXT0}M109 T0 S{TEMP0}

Printing works with the above G-Code.

My Guess as to what might be happening is below:
https://drive.google.com/file/d/15O22-1IO5VKS0Qhk81dHlJNffPyB0eHQ/view?usp=sharing

luc-github, 2019-01-02:
Hi IMHO your first approach was correct : here my start code for 3 years (I move down Z because of corner plates and not flat bed) :
```
M116;wait all temperature to be reached
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
G28 ; Home extruder
G1 Z15.0 ;move the platform down 15mm
G92 E0 ;zero the extruded length
G1 F200 E3 ;extrude 3mm of feed stock
G92 E0 ;zero the extruded length again
M100
G4 P2
;Indiana
M300 S1318 P240
M300 S0 P120
M300 S1396 P120
M300 S1567 P120
M300 S0 P120
M300 S2093 P960
M300 S0 P120
M300 S1174 P240
M300 S0 P120
M300 S1318 P120
M300 S1396 P960
M300 S0 P240
M300 S1567 P240
M300 S0 P120
M300 S880 P120
M300 S987 P120
M300 S0 P120
M300 S2793 P960
M300 S0 P240
M300 S880 P240
M300 S0 P120
M300 S987 P120
M300 S2093 P480
M300 S2349 P480
M300 S2637 P480
```

also looking at https://drive.google.com/file/d/1WeqQqY0UhzlpYkdpPPkKsiMyh_DTcyfr/view
it looks at home position, your Z in under 0 as nozzle is under bed - this mean you bed is too high , leveled but too high, a home, Z should be 0 not minus something.

I do not says there is no bug in Zmin evaluation calculation but at least your picture is clear - bed is too high a z home and modifying Z min is not the good approach as if you do a Z home when noozle is in middle of bed you can have issue.

Also be noted that Davinci beds are well known to not be flat but curved, so sides can be 'higher' than center that why it is a Zmin `evaluation` it is an average not a fixed value.
All that to say that the best solution for most of machine is to low down the bed when start print and have a Z = 0 at home position

sbeaman, 2019-01-02:
> it looks at home position, your Z in under 0 as nozzle is under bed - this mean you bed is too high

Yes, exactly.
- When you save the results of Zmin evaluation to EEPROM
- Issue a G28
- The bed is too high resulting in:
    - The Zprobe clearing the bed
    - The nozzle striking the side of the bed
- The cause of this is an error in Zmin evaluation. The bed is too high by the value equal to Z_PROBE_HEIGHT.

(Now if you mean the bed adjustment screws have the bed too high. No they are turned down as low as they can be set and still have Zmin evaluation touch them. Any further the probe stops before reaching the plate and an error is returned)

> I do not says there is no bug in Zmin evaluation calculation
- That is fine, but this a bug report about an apparent error in the Zmin evaluation based on the current value of Z_PROBE_HEIGHT that is causing the bed height issue.
- That is why I suggest a value for Z_PROBE_HEIGHT of 0.00 (When the Zprobe touches the calibration plate(s) the head is at bed level. https://drive.google.com/file/d/1pEgPHZj-atxbFrDWUbl3o6AFRGrLs1KS/view?usp=sharing)
- I verified this by measuring how far above each calibration point is above the bed. Each calibration point is raised above the bed by an equal amount as the zprobe if offset from the nozzle. (with in +/- 0,015)
- I have also verified setting Z_PROBE_HEIGHT to 0.0 corrects the issue (or at least a value of 0.00 - 0.03).

So if you:
 - Set Z_PROBE_HEIGHT to a value between 0.00 - 0.03. (I get the best printing results with a value of 0.03, but that may just be my system).
 - Save the result of Zmin evaluation to EEPROM
 - Issue G28
 - The bed is at the correct height. The nozzle clears perfectly. Printing works as expected.

luc-github, 2019-01-02:
The bed is too high means you need to use the screws to low down, changing zmin won t change bed position as end stop is always at same position

sbeaman, 2019-01-03:
Awww  G28 is always hard endstop. I incorrectly assumed it took ZMIN into account. I should of looked at the code before I made such an assumption. My mistake.

So Yes the nozzle hitting the bed at G28 is definitely a bed height issue.

closed 2019-01-03 by sbeaman

mdsandell, 2020-04-24:
I have this exact same issue and am unclear what you finally did to resolve it. Can you provide any help? Thank you!

sbeaman, 2020-04-28:
> I have this exact same issue and am unclear what you finally did to resolve it. Can you provide any help? Thank you!
> 
> **Update:** My z min position in EEPROM was < -2, so I set that to zero, and I set the z probe test distance to 10mm, and that allowed me to manually adjust the bed appropriately again.

I didn't. I wrote my own version of the ZMin calculation to change its behavior.

hyx0329, 2022-10-29:
For my davinci 1.0(JP1), I have applied the following configuration:

```cpp
#define Z_PROBE_HEIGHT 0.01 /* original 0.28, but on my machine, the minimum gap between
the nozzle and the bed when probing just fits a sheet of paper (80g/m^2) */
#define Z_PROBE_X1 -9  //original -8, the nozzle might hit the conductive plate
#define Z_PROBE_X2 -9  //original -8, the nozzle will hit the plate
#define Z_PROBE_X3 178  //original 179, the nozzle might hit the plate
```
I would suggest anyone flashing the firmware to take a look at these values to make sure they fit the actual machine.

Also, it's good to have zMin == 0, but not recommended on davinci 1.0 (feasible but useless and time-consuming unless you want to make sure the nozzle will never hit the bed). If it's a must, try to mod the sensor mount to make it adjustable, which introduces less pain.