luc-github/Repetier-Firmware-4-Davinci #334: Probleme etalonnage X/Y

GIRAUDGPH, 2018-12-29:
Bonjour
J'ai installe sans problème le Firmware Repetier sur ma DaVinci 1.0 AIO et tout fonctionne parfaitement sauf que je me suis rendu compte que les dimensions en X et Y n'étaient pas correct (0.3mm d’écart sur les deux axes).  
J'ai voulu corriger en modifiant le nombre de pas / mm des axes x/y dans l'EEPROM 
(80.68 steps/mm sur les deux axes) et la, l'axe Z est déréglé et le plateau ne se positionne plus correctement malgré la bonne calibration du départ.En remettant les anciennes valeurs (80 steps/mm) tout rentre dans l'ordre.
comment faire pour avoir les bonnes dimension des objets si on ne peut pas modifier ces valeurs?
si quelqu'un peut m'aider, merci d'avance.

luc-github, 2018-12-29:
modifier les step/mm de X et Y est la bonne facon de calibrer les axes X et Y et ne devrait pas affecter Z
[image]

Attention il y a aussi une marge d'erreur a prendre en compte si l'on utilise de l'ABS - l'ABS se retracte pas mal en refroidissant comparer au PLA

GIRAUDGPH, 2018-12-29:
merci pour cette réponse rapide
je n'utilise que de l'ABS et c'est justement pour obtenir un objet fini à la bonne dimension (une fois refroidit )que je voulais modifier les valeurs des axes X et Y .
Il est quand même étonnant que l'axe Z soit modifier et ne se positionne pas correctement pour la 1ere couche 3,27 mm au lieu de 0.27 mm (90% de 0.3 mm)
le Z min est bon ainsi que le nivellement du plateau.
lorsque je remet les valeurs par défaut (80 step/mm) tout rentre dans l'ordre sans rien modifier d'autre
et le plateau se positionne bien a z=0.27 et l'objet s'imprime correctement
je joint le début du Gcode généré par Slic3r 

M190 S100 ; set bed temperature and wait for it to be reached
M109 S220 ; set temperature and wait for it to be reached

; Filament gcode

M109 S220 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G92 E0
M106 S89.25
**G1 Z0.270 F7800.000**
G1 E-2.00000 F2400.00000
G92 E0
G1 X85.146 Y85.841 F7800.000
G1 E2.00000 F2400.00000

luc-github, 2018-12-29:
Oui je suis d accord c est bizarre 3.27 au lieu de 0.27...

2 choses a tester:
A - passer les commands manuellement
G28 puis G21 puis G90 puis G1 Z0.270 F7800.0
Il manque G28 dans le start code, c est normal?
B - voir le changement qui genere le probleme
1 - modifier X seulement et8 attendre 30s puis redemarer l imprimante et tester
2 -  modifier Y seulement et attendre 30 s puis redemarrer l'imprimante et tester

Pour info cet etalonnage devra etre fait pour chaque nouvelle bobine de filament car chaque filament reagit differement a la contraction

GIRAUDGPH, 2018-12-29:
Merci encore pour votre disponibilité 
J'ai constaté que même issue du même fournisseur les caractéristiques  des filaments diffères
Je n'ai plus le temps ce soir pour tester vos suggestions mais des demain matin je teste et je vous tiens au courant 
Merci encore

luc-github, 2018-12-29:
oui chaque filement a une composition differente meme venant du meme fourniseur ^_^ les joies de l'impression 3D

GIRAUDGPH, 2018-12-30:
Bonsoir
après maintes essais, j'ai enfin trouvé la raison de mon problème
J'utilise plusieurs configuration d'EEPROM (.epr) et celui que j'utilisai pour mes tests de calibration contenait des valeurs négatives dans les positions de parcage des axes X et Y. J'envoi un M100   dans le script de démarrage de l'imprimante (au niveau des paramètres imprimante de Repetier-Host) ce code descend Z de 3 mm et lance la séquence de nettoyage de la buse. 
Je pense que les valeurs erronées des axes X /Y dan l'EEPROM empêche Z de se repositionner au bon niveau (0.27) au démarrage de l'impression.
Envoyer un G28 manuellement m'a permis de voir que les Valeurs de X et Y étaient en rouge sur le tableau de bord des contrôles manuels de repetier et que l'axe Z ne bougeait pas.
merci de votre aide précieuse et bonnes fêtes de fin d'année.

luc-github, 2018-12-30:
bien - Merci et bonnes fetes ^_^

closed 2018-12-30 by luc-github