luc-github/Repetier-Firmware-4-Davinci #335: Using manual bed level after changing nozzle for stock 1.0a

moorer2k, 2019-01-17:
First off thanks for your hard work, always appreciated it.

So, I read similar issues but nothing really seemed to provide the exact answers for me. I recently purchased this product [XYZ Printing Da Vinci 1.0 Pro MK10 M7 Nozzle Adapter](https://www.etsy.com/listing/594312121/xyz-printing-da-vinci-10-pro-mk10-m7) for my 1.0a. Installed the z bracket for hitting the sensor, since it's about 11mm longer than the stock setup was.

When I home all the axis's, the nozzle is nice and clear from the the bed. When I go to manually level the bed, it raises the bed much higher than what the endstop is triggering at.. If I understand correctly, this has changed the standard z height defined in the firmware by AT LEAST -11mm (the offset the new nozzle has added) ? If I should ask this @ repetier forum I will head over there, and sorry to post at the wrong location.

moorer2k, 2019-01-17:
Yeah I make sure when I home all axis's that the nozzle is off to the side! But, when manually leveling bed, it starts to come up, and it keeps going up and pushes really hard on the nozzle and i have to power it off. Should I change the Z_MAX_LENGTH to 189 or so and see if that helps with manual bed level or would that mess anything else up?

moorer2k, 2019-01-17:
Yes I understand that part. I guess what I am asking is where would I need to look to check why the manual bed leveling is causing the bed to lift into the nozzle. After homing all, I disabled the steppers and moved the extruder over the bed and it has nice clearance so I’m not sure if it’s an offset I need to input into repetier or? Here’s some pics, just for fun :)
 
**new nozzle adapter**
[image]

**z bracket for extended end stop triggering**
[image]

luc-github, 2019-01-17:
Then be sure in EEPROM Zmin is 0 
[image]

as well in your slicer if set

moorer2k, 2019-01-17:
I also forgot to mention that I did not reinstall the z probe, is this pretty critical? It came with a new z probe to install, could this be the problem for the manual bed leveling? My z min is 0 yes.

moorer2k, 2019-01-17:
[image]
[image]
[image]

luc-github, 2019-01-17:
for manual bed levelling z probe is not used but yes do not use autobed leveling

moorer2k, 2019-01-17:
Ok so I guess I was wrong... something must of reset a setting or something... it actually isn’t homing in the right position at all anymore. I was really tired {bare with me, I’m a narcoleptic, so much is like a dream to me}. When I home it’s now off even with the new bracket... I loaded the eeprom failsafe just to start fresh and make sure it’s stock. This what I’m dealing with distance wise now after homing.

[image]

moorer2k, 2019-01-17:
So if I set z min to 11, then home all, it the bed moves down to where I can clear it fine. But that won’t stay in effect when I do manual bed leveling. 
[image]
[image]

moorer2k, 2019-01-17:
Ok I’ll give it a shot and let you know by the end of the day the outcome !

moorer2k, 2019-01-18:
Ran into some more issues unfortunately... The PCB for the extruder seemed to have some bad pins for the thermistor, even the fans seems like they are both trying to break apart. One has severed completely off the board. I think i'll solder the fans directly to the the pins, and find a different way to connect the thermistors sensor back. Thanks again though for the information!

closed 2019-01-18 by moorer2k