luc-github/Repetier-Firmware-4-Davinci #336: Filament change problem with SD printing

arpruss, 2019-01-23:
When my 1.0a runs of filament in printing SD, and I load new filament using the Maintenance menu, for the next minute or more, the printer prints without extruding filament.

The filament extrudes just fine when loading with the Maintenance menu, but then when the head moves back in position, it stops extruding. I tried pushing the filament in by hand in case it was stuck, and I got a definite feeling that the extruder motor was not operating.

Here is a guess--just a guess. Maybe the printer's filament position gets changed by the extrusion happening during the Maintenance load operation, and the g-code extrusion position is no longer synchronized with the filament position? (But that wouldn't explain why I just had the problem in relative extrusion mode, too.)

luc-github, 2019-01-23:
Nothing changed for years now but it may possible
what version of fw do you use ? - check printer menu in settings-. system version

also what steps need to be done to reproduce this issue?
1 - SD printing
2 - press pause ? or automatic out off filament ?
3 - etc...

arpruss, 2019-02-14:
1. SD printing
2. press pause
3. maintenance: unload filament
4. maintenance: load filament, press OK to feed more until new filament is feeding
5. press resume

The new filament doesn't begin feed for quite a while after this--about a whole layer on a small print.

luc-github, 2019-02-14:
What the fw version ?

arpruss, 2019-02-14:
I grabbed it from your github Nov. 12, 2017.

arpruss, 2019-02-14:
I notice that there is no problem if I switch filament manually, without using the panel functions, by using the extruder release lever to remove the old filament and then pushing the new filament into position while holding down the release lever. After that, it prints just fine.

luc-github, 2019-02-15:
Ok understood

closed 2019-07-11 by stale[bot]