luc-github/Repetier-Firmware-4-Davinci #337: Da Vinci 1.0A heater/sensor failure and no head temperature after flash

MyPetGoomba, 2019-01-27:
I managed to flash my Da Vinci 1.0A with your latest Repeater firmware. In the Configuration.h I defined DAVINCI as 1 (for 1.0) and MODEL as 1 (for the J37 jumper). With the new firmware flashed, I can move the printing bed using either the Repetier Host or the printer buttons, but I can't print. The extruder always reads as 0 degrees and I get a constant heater/sensor failure warning that locks all my heaters. The bed reads as room temperature. 

I unplugged the extruder's thermister and tested it. It was about 116kohms (should be 100k, I believe). I tested the J4 fuse on the main board, as well. it isn't blown. I'd think it was a hardware issue if it wasn't for that and everything working fine with the stock firmware. Now I'm leaning towards a pin being set wrong somewhere. Could I get any help? I know I'm late to this game.

Thanks in advance.

Maturimu, 2019-01-27:
Did you do a "load eeprom default settings" after updating the firmware?

luc-github, 2019-01-27:
Please clarify what you call : `latest Repeater firmware`
What is the branch you used ? master ? devt ? 
To check pin is ok  :
If temperature is 0 when not heating what is the temperature when you disconnect sensor?
Does temperature change  if you use an hair dryer to heat your noozle ?

MyPetGoomba, 2019-01-27:
The error wouldn't let me send M502 or M500 from Repetier Host, but I used "Load Fail-Safe" from the printer menu and that seemed to work. 

I used the Master branch.

Temp reads 0 degrees when thermistor is disconnected. Interesting results with the hair dryer, though. When I heated the nozzle with one, the temperature rose past zero and I was then able to clear the error and gain control of the printer. I stuck a multimeter in there and took some temperature measurements. Made a table of what the printer displayed and what my multimeter displayed.  Don't trust the multimeter too much, but this table would explain why it thinks room temperature is 0 degrees. It's already displaying more than 20 below at 54 degrees.

displayed | measured | 
      30      |       54       |  
40 | 61 |  
50 | 68 |  
60 | 71 |  
70 | 77 |  
80 | 85 |  
90 | 93 |  
100 | 94 |  
110 | 100 |  
120 | 105 |  
130 | 111 |  
140 | 118 |  
150 | 125 |  
160 | 140 |  
170 | 145 |  
180 | 156 |  
190 | 159 |  
200 | 156 |  
201 | 156 | stopped here and wouldn't go higher
(When the thermistor was unplugged, the printer reading dropped right back to 0 from 40.)

Now, I'm new to this whole business, so forgive me if I just ask what this means. 
I know there is a thermistor table in the configuration.h. The table I pasted above was made with "#define EXT0_TEMPSENSOR_TYPE 1", but I first ran into the error after flashing the printer with this still defined at the default 5. 

Oh, and thanks for responding so fast. I posted only last night! (U.S. Central Time)

luc-github, 2019-01-27:
Do not change the default thermistor tables, they are experimental and the others ones won't work on Davinci printer because they are for another hardware, you can have a look to this FAQ : https://github.com/luc-github/Repetier-Firmware-4-Davinci/issues/104

so keep `#define EXT1_TEMPSENSOR_TYPE 5`
which correspond to :
`#define USER_THERMISTORTABLE0 {{294, 2560},{375, 2400},{461, 2272},{526, 2184},{601, 2104},{674, 2048},{811, 1928},{925, 1848},{1071, 1768},{1219, 1696},{1445, 1584},{1673, 1496},{1902, 1416},{2187, 1312},{2454, 1224},{2672, 1152},{3000, 1048},{3196, 968},{3439, 864},{3583, 784},{3736, 680},{3899, 536},{3960, 448},{4024, 336},{4067, 192},{4079, 156},{4100, 0},{4120, -160}}`

it was provided by people having 1.0A - it may not be fully fitting your printer - Davinci printer has higher tolerance in error for component compare to standard printer which make the measurement little bit hard to match all printers.
I do not have 1.0A so I cannot help a lot on values

MyPetGoomba, 2019-01-28:
Well, I performed my original steps over again with the tempsensor type back on 5. This time it worked flawlessly. Not sure what I did differently. Maybe Maturimu was right and I misclicked the the eeprom settings on my first attempt. Not sure, but it was probably my fault. 

Thanks a bunch. Your dedication is admirable.

Maturimu, 2019-01-28:
BeeRated, I can assume that the transistor controlling this heater is broken. Check it, if it is broken, you can replace it or desolder it off and give a control signal to something more serious, like this (fine fo hotend too, add fuse):  https://www.amazon.com/Printer-Heating-Controller-Heatbed-Extruder/dp/B075FCZQYT/ref=sr_1_8?ie=UTF8&qid=1548658191&sr=8-8&keywords=3d+printer+heatbed+control+module
It is possible that the heater could burn out and feed + to the case, I would suggest checking the contact between the heater leads and its case before turning on something

luc-github, 2019-01-28:
Davinci hardware is super fragile. The new generation is even more.

I guess issue can closed

closed 2019-01-28 by luc-github