luc-github/Repetier-Firmware-4-Davinci #340: Mac specific instructions

DavidLQuick, 2019-02-11:
I have read through the instructions and in step 4 it says to update the hardware libraries. All of the on line resources show people replacing the Arduino libraries on a windows machine using explorer. There are no similar resources on how to do this for the Mac version of the IDE. Is this even necessary on a Mac. If not can the docs be updated to say that?

luc-github, 2019-02-12:
the instructions is not OS specific it is for all OS, as Davinci need specific variant.cpp what ever the OS.
 
About how to do on Mac, there is a link to FAQ how to find the file locations what ever the OS: https://github.com/luc-github/Repetier-Firmware-4-Davinci/issues/114#issuecomment-231949044

DavidLQuick, 2019-02-12:
> About how to do on Mac, there is a link to FAQ how to find the file locations what ever the OS: #114 (comment)

Luc, 
I compiled today without any issues. I did not update the referenced files yet. I had already read the FAQ 114.

luc-github, 2019-02-12:
Compilation will always be fine. But without updating variant.cpp you will have black bars in screen after flashing

DavidLQuick, 2019-02-12:
Thanks, I’m snowed in, in Seattle so I'm working on it tomorrow.

DavidLQuick, 2019-02-12:
Done, and now to bite the bullet and convince myself I can do without the printer if it doesn't work. Tomorrow maybe.

luc-github, 2019-02-12:
There is no reason it won t work for you. It works for others users

DavidLQuick, 2019-02-16:
Is there a guide on setting the flashed printer up to print? The instruction say to calibrate your Z from the menu but no detail on how to do that.

luc-github, 2019-02-17:
Check on google I am sure there are several walkthrough - now you are flashed the printer is like any other 3D printer 
I guess you mean :
`When update is complete you must calibrate your bed height!Use manual bed leveling in menu`
which is different than `calibrate your Z`

well you have a a menu ` manual bed leveling` click on it and follow instructions on screen, and if still not clear: https://www.google.com/search?q=manual+bed+leveling+3d+printer

DavidLQuick, 2019-02-17:
Got it running. Printing a Benchy. First thing I noticed is it is way quieter. Thanks for any help.

luc-github, 2019-02-17:
Good now you have a free to mod printer - have fun - happy printing

closed 2019-02-17 by luc-github