luc-github/Repetier-Firmware-4-Davinci #341: Unable to Flash Da Vinci 1.0A using repetier

CollinsCreations1, 2019-02-12:
Sorry for the double post, I believe I commented on a closed section previously and this is my first time using GitHub. Hello I've been having ALOT of trouble bouncing between forums and videos figuring out the hard way as to which method is outdated. I am trying to flash my XYZ Da Vinvi 1.0A printer with repetier host software so that I may print with 3rd party filaments and use the Cura slicer. I've tried every Arduino version including 1.8.0 and have always added the correct Board through manager and selected the native USB option. I have selected COM 4 as this is the printer connection and have entered 1 for both of the configuration options. I started by flashing the printer by shorting the J37 pins and achieved the 2 black bars. I have transferred the correct hardware files from the repetier folder into the arduino folder. When I verify the arduino code it says successful however every time I upload to the printer even if it results in the code saying "upload complete/successful" the USB becomes unrecognized by my computer and the only way to re-establish the printer as "COM 4" again is by reflashing by shorting J37 again. It's been an endless loop of failure for me and I can't quite find the solution. As I've read from above there is a "Bossa Procedure" however I can't quite seem to find anything to walk me through it. If anyone has any recccomendations, I'd very much appreciate it!

luc-github, 2019-02-12:
This is typical issue of variant.cpp no copied in proper location as described on FAQ
so original is still used.
What is the file location of the variant.cpp you updated ?
Give me full path pls

CollinsCreations1, 2019-02-12:
I copied the file from the repetier folder found at

 C:\Users\colli\Desktop\Repetier-Firmware-4-Davinci-master\src\ArduinoDUE\AdditionalArduinoFiles\Arduino - 1.8.0 -Due 1.6.8\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x

to the arduino folder found at 

C:\Users\colli\Desktop\arduino-1.8.0\hardware\arduino

I noticed that the arduino file doesn't seem to contain the variant.cpp file orginally, so only the usbcore is replaced and the variant seem to be added rather than replaced. Sorry if i am wrong. Also I have posted 2 screen shots. The first shows where I copy the variant file from. the second displays that I do not have a file labeled "sam" to paste the variant into, therefore I can't get to the arduin due x file. 
Thank you so much for your feedback!
[image]

[image]

luc-github, 2019-02-12:
You need to download due module first like explained in readme step 3

CollinsCreations1, 2019-02-12:
Is the step 3 pasted below you are referring to? If so, I posted a picture showing I have installed the due 1.6.8 from tools/board manager
3. Use an arduino IDE supporting arduino DUE, 1.8.0 with Due 1.6.8 module from board manager FAQ#166.
[image]

CollinsCreations1, 2019-02-12:
I have moved the variants file to this location, does this look correct? (Sorry I can't test it myself I am at the library but will test when I am home in 2 hours if it is the correct format?)
C:\Users\colli\Desktop\arduino-1.8.0\hardware\arduino\sam\1.6.8\variants\arduino_due_x
[image]

luc-github, 2019-02-12:
The target file is a replacement. So if no fale is not correct location. And Desktop is not arduino location. Did you follow step 4 Faq link ? It open right location

luc-github, 2019-02-12:
FAQ#114

CollinsCreations1, 2019-02-12:
Okay, I believe I transferred the files as they should be. I will update you when I am able to test this out on my printer in about an hour. Thank you so much for all the help so far!

CollinsCreations1, 2019-02-12:
I cannot thank you enough, the printer booted into it's new firmware, thank you for helping me so patiently and sorry I was essentially reading over the obvious instructions and was caught in my habit for saving things to the desktop haha.

closed 2019-02-12 by CollinsCreations1