luc-github/Repetier-Firmware-4-Davinci #342: Extruder crashing into bed

DavidLQuick, 2019-03-11:
I converted to a Bondtec and E3d v6. I home the system. I used the panel position controls to manually move the print head over the center of the bed and the raised the bed to where it just drags a piece of paper.

 I look at the z value 2.20 mm. So what do I change to make the bed always go to this position before starting a print. 

I have tried setting the z offset to 2.20, and alternatively setting z to this number but when I home it reverts the offset to 0 and in the second case crashes the probe into the sensor.

DavidLQuick, 2019-03-11:
I set the vault in bed coating and it is working but I can’t believe this is the right place.

luc-github, 2019-03-11:
same issue as https://github.com/luc-github/Repetier-Firmware-4-Davinci/issues/333#issuecomment-451032057
What you want to do is changing zmin to -2.2mm so when printing set bed to position 0.2 will be correct
but this is not the right approach, when homing Zmin must be  0 because it is hardware end stop not software end stop, so if you home Z even you already set Zmin is -2.2 nozzle will crash bed.

the solution is to redo your bed leveling using screws

DavidLQuick, 2019-03-11:
> same issue as #333 (comment)

I don’t think there is 2 mm of travel left in the screws bu I’ll try.

luc-github, 2019-03-11:
In that case you need to move the endstop  or to put the flag higher like here https://www.thingiverse.com/thing:621440

DavidLQuick, 2019-03-11:
The original thing for the E3d had such a device. They also said they lost 23.5 mm of build volume.  I figured since my design places the hot end at almost the same place as stock it should not be needing such a thing.

Got to do something about that wiring.

This is the whole reason for moving to Repetier.

luc-github, 2019-03-11:
2.2 mm is not that big , hope you can handle with the screws only

luc-github, 2019-03-18:
I guess issue can be closed

closed 2019-03-18 by luc-github