luc-github/Repetier-Firmware-4-Davinci #343: Winstar LCD on a DaVinci serial number starting with 3F10A

curnowdj, 2019-03-18:
I can not get the Winstar LCD on a DaVinci serial number starting with 3F10A working. The upload seems to go fine but no joy with the LCD. It has the back light and the 2 black bars but no characters.
This my config.h
/*
    This file is part of Repetier-Firmware.

    Repetier-Firmware is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    Repetier-Firmware is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with Repetier-Firmware.  If not, see <http://www.gnu.org/licenses/>.

*/

#ifndef CONFIGURATION_H
#define CONFIGURATION_H

// ################## EDIT THESE SETTINGS MANUALLY ################

#define DAVINCI 2// "0" if not DAVINCI, "1" For DAVINCI 1.0, "2" For DAVINCI 2.0 with 1 FAN, "3" For DAVINCI 2.0 with 2 FAN, 4 for AiO (WITH NO SCANNER SUPPORT)
#define MODEL  1//"0" for first generation (jumper JP1 to reset ) , "1" for new generation   (jumper J37 to reset)
#define REPURPOSE_FAN_TO_COOL_EXTRUSIONS 0 //Setting this to 1 will repurpose the main Extruder cooling fan to be controlled VIA M106/M107
                                           //Warning: for DaVinci 1.0 need to add a permanent fan with power supply to cool extruder

// ################ END MANUAL SETTINGS ##########################

//Version
#define VERSION_MAJOR "1"
#define VERSION_MINOR_YEAR "17"
#define VERSION_MINOR_MONTH "08"
#define VERSION_MINOR_DAY "01"
#define VERSION_BUILD "1"

//Davinci screen is not standard reprap it is WINSTAR 16x4
#if DAVINCI==0
#define WINSTAR_SCREEN 0
#else
#define WINSTAR_SCREEN 1
#endif

//Decouple Test feature, do not disable it unless you know what you are doing!!!
//if you have decouple issue it means you have hardware issue or bad air flow management!!!
//this feature is a safety feature, if you disable it is dangerous!!!
#define FEATURE_DECOUPLE_TEST 1
//this will hide on sd card no extension files and bin/hex/dat files to make navigation and selection easier
#define HIDE_BINARY_ON_SD 1
#define UI_AUTOLIGHTOFF_AFTER 1
#define ENABLE_CLEAN_DRIPBOX 1
#define ENABLE_CLEAN_NOZZLE 1
#define FEATURE_ENCODER	0
//ensure of some define if AiO
#if DAVINCI==4
//no drip box
#undef ENABLE_CLEAN_DRIPBOX
#define ENABLE_CLEAN_DRIPBOX 0
//it must be model 1
#undef MODEL
#define MODEL 1
#endif
//to enable communication using wifi module set to 1
#define ENABLE_WIFI 0
//define the wifi serial output
//on Davinci use Serial
//on RADDS use Serial1
[... 1995 lines cut ...]

/*
#define USER_KEY1_PIN     UI_DISPLAY_D5_PIN      // D5 to display (not used for graphics controller), change to other pin if you use character LCD !
#define USER_KEY1_ACTION  UI_ACTION_FAN_SUSPEND
#define USER_KEY2_PIN     UI_DISPLAY_D6_PIN      // D6 to display (not used for graphics controller)...
#define USER_KEY2_ACTION  UI_ACTION_SD_PRI_PAU_CONT
#define USER_KEY3_PIN     UI_DISPLAY_D7_PIN      // D7 to display (not used for graphics controller)...
#define USER_KEY3_ACTION  UI_ACTION_LIGHTS_ONOFF
#define USER_KEY4_PIN     -1
#define USER_KEY4_ACTION  UI_ACTION_DUMMY
*/

// ####### Advanced stuff for very special function #########

//#define NUM_MOTOR_DRIVERS 0
// #define MOTOR_DRIVER_x StepperDriver<int stepPin, int dirPin, int enablePin,bool invertDir, bool invertEnable>(float stepsPerMM,float speed)
//#define MOTOR_DRIVER_1(var) StepperDriver<E1_STEP_PIN, E1_DIR_PIN, E1_ENABLE_PIN, false, false> var(100.0f,5.0f)

/*
  You can expand firmware functionality with events and you own event handler.
  Read Events.h for more informations. To activate, uncomment the following define.
*/
//#define CUSTOM_EVENTS

#endif

luc-github, 2019-03-18:
Yes issue is with variants.cpp
Did you read the FAQ ?

luc-github, 2019-03-18:
on linux it is as described in FAQ
[image]

curnowdj, 2019-03-18:
Yes I know where to find variants.cpp but where should it go in the arduino folders?

luc-github, 2019-03-18:
what you mean ? did you look at screenshot ? this is arduino folder

curnowdj, 2019-03-18:
So the whole sam folder goes in the arduino hardware folder?

luc-github, 2019-03-18:
The read me say :
Update variants.cpp/USBCore.cpp arduino files with the ones present in src\ArduinoDUE\AdditionalArduinoFile according your IDE version.

and the files are from my repo /src/ArduinoDUE/AdditionalArduinoFiles , take ones according the IDE version

curnowdj, 2019-03-22:
OK I think I have that sorted but now I can not get the usb port to connect. Should I start a new issue or continue here?

[image]

luc-github, 2019-03-22:
you mean printer is working by modifying the variant.cpp you screen shot ?- no black bars and you can navigate in menu ?
I have serious doubts ...

curnowdj, 2019-03-22:
No. Printer not working. Black bars on lcd so no fw. But now I can not connect via usb. I did a search for this issue and I suspect the main board has failed, so it looks like I will go back to my original plan and go down the Mega Ramps road. Thanks for you help though.

luc-github, 2019-03-23:
It is same issue you use wrong variant https://github.com/luc-github/Repetier-Firmware-4-Davinci/issues/172
You need to use jumper  to recover

curnowdj, 2019-03-23:
No point doing jumper if I can not connect via the usb.

luc-github, 2019-03-23:
Doing the  jumper allow you to connect again...

curnowdj, 2019-03-24:
No. Even after doing J37 I still  can not connect via USB.

luc-github, 2019-03-24:
sometimes  need to do several times to wipe fully the FW

curnowdj, 2019-03-24:
OK I have done the jumper a couple of times. I will redo it a few more times and see if it comes good.

curnowdj, 2019-03-31:
No update yet. I'm having a break from it due to the frustration.

luc-github, 2019-03-31:
Ok so I close issue for now - please re-open if necessary.
There is no reason you cannot flash like others did - the only possibility is the motherboard is damaged but it can not be done by flashing wrong variant.

closed 2019-03-31 by luc-github