luc-github/Repetier-Firmware-4-Davinci #344: M501 getting back M92 Values

celticht32, 2019-04-12:
When I enter in the M501 command from the command prompt in either the web server or from  Repetier Host I do not get anything back or I get the following (connecting from repetier host)

17:25:20.953 : Repetier-Server successfully connected.
17:25:21.265 : M115
17:25:21.531 : FIRMWARE_NAME:Repetier_0.91 FIRMWARE_URL:https://github.com/repetier/Repetier-Firmware/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:Mendel EXTRUDER_COUNT:1 REPETIER_PROTOCOL:2
17:25:21.531 : Printed filament:636.80m Printing time:4 days 4 hours 39 min
17:25:32.430 : M501
17:25:32.540 : Info: 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000
17:25:33.243 : Info:Configuration loaded from EEPROM.ok 11080
17:25:34.165 : Warning: Missed line detected - correcting buffer usage.

I do not get anything back with 👎 
M92 X80.2 Y80 Z2044 E104

I changed my belt on my davinci 1.0a and now the X axis is way off on its  movements..

I selected X access to move 100mm and it only moved 81.78mm...

So I am trying to figure out what the current M92 value is for the base level code..
I went into config.h and found the following that I am assuming is the M92 base level:

/** Drive settings for printers with cartesian drive systems */
/** \brief Number of steps for a 1mm move in x direction.
For xy gantry use 2*belt moved!
Overridden if EEPROM activated. */
#define XAXIS_STEPS_PER_MM 98.425196

So Taking the formula:
M92 X = Desired Movement / Actual Movement * Current M92

I get = 100/81.76*98.425196 = 120.38

I then do M92 X120.38 to fix the steps.

I then do a M500 and it says its stored to eprom...
do a M501 and get the following:

17:35:57.466 : M92 X120.38
17:36:04.903 : M500
17:36:05.997 : Info:Configuration stored to EEPROM.ok 11712
17:36:06.325 : Warning: Missed line detected - correcting buffer usage.
17:36:17.122 : M501
17:36:17.215 : Info: 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000
17:36:17.325 : Info:Configuration loaded from EEPROM.ok 11725
17:36:18.465 : Warning: Missed line detected - correcting buffer usage.
So then I do another test... and it is still wrong on the X axis...

1. How to I change the Steps per MM for the X axis since it appears that M92 and M501 do not work correctly.
2. Can this be added to be fixed?

celticht32, 2019-04-12:
M92 is working but M501 is returning nothing but what you see above.

closed 2019-04-12 by celticht32

luc-github, 2019-04-13:
1 - first you use a very old version as I see you use 0.91 based which is not for 1.0a but I see you can print 4days  so I am confused of what FW you use and what is your printer
2 - M501 load FW setting to memory, when loading it check so value but do not display them
here normal output with 0.92.10 based FW for `M501`
```
SelectExtruder:0
FlowMultiply:100
Info:Configuration loaded from EEPROM.
```
3 - Modifying settings then saving them to EEPROM with `M500` is enough, you do not need to load EEPROM again, actually you never need to use `M501`, EEPROM is read when printer start automaticaly.

4 - if you want to display EEPROM settings you must use `M205` this command display EEPROM settings