luc-github/Repetier-Firmware-4-Davinci #347: Extruder 1 (Right) won't print unless 0 (left) is also printing

Shawnh77, 2019-04-29:
When slicing and printing I select the right extruder (ext1).  Everything looks correct in the print preview.  However, after the bed heats up the head never does and it will go through all the motion of printing, but obviously never extrudes and filament.  If I preheat for ABS and then print an object with two colors everything goes fine (the traffic cone for example).  If I try to only use the right extruder I cannot get anything to print.

Any ideas what I've done wrong?

luc-github, 2019-04-29:
Did you verify your generated GCODE ?

Shawnh77, 2019-04-30:
I have to admit that I am not very good at identifying the error. I'm guessing the T0 and T1 are the extruders switching but the right extruders never feeds.  Here is a copy of the code(I removed some to fit within limits):

[canisters.txt](https://github.com/luc-github/Repetier-Firmware-4-Davinci/files/3131290/canisters.txt)

luc-github, 2019-04-30:
T0/E0 is left extruder T1/E1 is right extruder
your GCODE only use left extruder (T0) so of course right extruder (T1) is not working

Shawnh77, 2019-04-30:
I'll go back through the options in Cura to see if I can figure out why.  It see it correctly in the Object Placement tab and in the print preview.

Shawnh77, 2019-04-30:
[image]

[image]

[image]

[image]

It is really strange, all the settings appear correct to me.  The print preview even shows the estimated amount of filament needed.

luc-github, 2019-04-30:
you need to check your start code - it have macro, by default if I remember well  (I do noit use cura since 5 years) start code handle only one extruder - need to be modified for a second one

Shawnh77, 2019-05-01:
I tried this for my start code.  The printer moved the heads as though it was printing but did not actually feed or heat the ext1

; Default start code
G28 ; Home extruder
G1 Z15 F{Z_TRAVEL_SPEED}
M107 ; Turn off fan
G90 ; Absolute positioning
M82 ; Extruder in absolute mode
{IF_BED}M190 S{BED}
; Activate all used extruder
{IF_EXT0}M104 T0 S{TEMP0}
{IF_EXT1}M104 T0 S{TEMP0}
G92 E0 ; Reset extruder position
; Wait for all used extruders to reach temperature
{IF_EXT0}M109 T0 S{TEMP0}
{IF_EXT1}M109 T0 S{TEMP0}

luc-github, 2019-05-01:
the error is pretty obvious if you read my previous comment
>T0/E0 is left extruder T1/E1 is right extruder

in 
```
{IF_EXT0}M104 T0 S{TEMP0}
{IF_EXT1}M104 T0 S{TEMP0}
```
and
```
{IF_EXT0}M109 T0 S{TEMP0}
{IF_EXT1}M109 T0 S{TEMP0}
``` 
use only T0 (left)
correct start code is :
```
{IF_EXT0}M104 T0 S{TEMP0}
{IF_EXT1}M104 T1 S{TEMP1}
```
and
```
{IF_EXT0}M109 T0 S{TEMP0}
{IF_EXT1}M109 T1 S{TEMP1}
```

Shawnh77, 2019-05-01:
You are absolutely correct.  I was focused on the EXT.  I am sorry to bother you.  Thank you for your help!

luc-github, 2019-05-01:
have fun

closed 2019-05-01 by luc-github