luc-github/Repetier-Firmware-4-Davinci #348: Davinci duo 2.0a j37 not extruding

Raymond41, 2019-06-10:
I recently flashed my Davinci Duo 2.0a printer with Repetier 0.92 firmware but, I have not been able te print anything since it does not extrude. It does extrude however in manual mode but when printing it does not extrude. It is not in dry mode. Does anyone has any suggestions?

luc-github, 2019-06-10:
if it extrude in manual mode - but not when printing - issue is with your slicer

Raymond41, 2019-06-10:
Alright, so it one of the settings then?

luc-github, 2019-06-10:
yes be sure you use right extruder in slicer, T0 is left extruder T1 is right extruder
be sure also you heat the right one.

check your generated GCODE to control what you get

if all is correct then check you bed leveling - to be sure you are not too close and so stop extrusion

Raymond41, 2019-06-10:
At first I thought the extruders were mixed up but after heating both up and trying to print I found out that was not the case. I can't seem to find the extruder settings in the Cura slicer, I mean selecting which is which, as you suggested.

luc-github, 2019-06-10:
I do not use cura for years, sorry - cannot help on this
I just can say that if no extruder is defined T0 / left one is used

Raymond41, 2019-06-10:
From a binary standpoint you'd think that T0 would be on the right side. Thank you for responding so quickly. I'll be fiddling around with the slicer then. If I come up with a solution I will let you know.

luc-github, 2019-06-10:
T0 is left extruder, T1 is right extruder this is how firmware is designed.

Raymond41, 2019-06-10:
I did check out the G-code though, it sends commands to extruder 0, the left one, so the correct one is heating up.

luc-github, 2019-06-10:
so in GCODE only T0 is used ? if yes as said it can be the bed leveling - extruder is too close and cannot extrude.
To check. change in your GCODE the Z position to lower bed brfore extrusion and see if extrusion is working.

Anyway if manual extrusion is working is not FW issue - I just can guess it is Slicer or bed leveling

Raymond41, 2019-06-11:
You were right, the extruder was too close and could not extrude. I did level the bed in the davinci mode before flashing so I thought it was oke, but i was wrong. It is now working fine so I like to thank you very much!

closed 2019-06-11 by luc-github