luc-github/Repetier-Firmware-4-Davinci #349: Davinci 1.0 Pro 3in1 Flashed, Bossa COM port not showing up

Captchaaa, 2019-07-09:
The issue as it stands:

I was updating my DaVinci firmware from 1.1.3 to 1.1.9, however, I've had a power issue which caused my printer to display black bars, and won't work at all. I didn't have to jump any pins or stuff that's been stated on other forums.

I thought maybe I can try and update it using Repetier Firmware like how other people are doing it, yet I'm having an issue where the bossa COM port isn't showing up at all under device manager.

Of course, when the printer's bootmanager launches(holding the up and down keys) it shows up as a port for the Davinci Jr, not the Pro 3in1 but in this state, I cannot update the drivers to that of a bossa port.

I've been told that bricking the printer is practically impossible, yet I can't seem to find a solution to this issue.

luc-github, 2019-07-09:
Davinci pro is not supported it is state in readme and FAQ. It is different board and have some hardware that make it incompatible with opensource firmwares
You confirm you tried to flash a Davinci pro ?

Captchaaa, 2019-07-09:
Thank you for the response luc,
I didn't intentionally flash the Davinci pro, though given that it now displays black lines, and the lights don't turn on leads me to believe I might've on accident. As of right now, I'm currently trying to find a way to fix it where it would print again.

luc-github, 2019-07-09:
I  am sorry to say that the only solution is to contact xyz support.

Captchaaa, 2019-07-09:
That's alright, Thanks anyway!

closed 2019-07-09 by Captchaaa