luc-github/Repetier-Firmware-4-Davinci #351: Error compiling.

danny1231233, 2019-08-02:


luc-github, 2019-08-02:
Did you read the readme ? It is wrote in bold that what you do is not correct

danny1231233, 2019-08-02:
So what do I have to do?

luc-github, 2019-08-02:
Read the readme. Follow the readme instructions.

jnowak1054, 2019-08-02:
Had trouble compiling for a while as well, this is what worked for me on Windows10 x64

1) Working with Arduino 1.6.9

2) In Arduino boards manager install Arduino SAM Boards(32-bits ARM Cortex-M3) version 1.6.8 (very important correct version is used)

3) Download and unzip the Repetier.92 software from https://github.com/luc-github/Repetier-Firmware-0.92 and unzip into a working folder

4) Copy the following Repetier file -
Repetier-Firmware-0.92-master\src\ArduinoDUE\AdditionalArduinoFiles\Arduino - 1.6.9 -Due 1.6.8\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x\variant.cpp
to
C:\Users\your user\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x\variant.cpp

5) Open the Repetier.ino, change the Board and Port (Board:"Arduino Due(NativeUSB Port)"), compile and upload the code

danny1231233, 2019-08-02:
for the da vinci 2.0A duo should i use https://github.com/luc-github/Repetier-Firmware-4-Davinci or https://github.com/luc-github/Repetier-Firmware-0.92

luc-github, 2019-08-02:
they are same

danny1231233, 2019-08-02:
what are the odds of this ruining my printer forever? This is my first time flashing anything?

luc-github, 2019-08-02:
if you do not short anything - none

danny1231233, 2019-08-02:
i do all of the steps correctly but when the file finishes uploading nothing happens to the printer and windows recognizes it as an unknown port.

luc-github, 2019-08-02:
where do you copy the variant.cpp file ?

danny1231233, 2019-08-03:
i added the variant file and now the upload of the firmware to the printer is stuck at 69%. how long should the upload process take?

danny1231233, 2019-08-03:
I got the firmware to install but the lights wont come on and the extruders wont move.

luc-github, 2019-08-03:
You can navigate in menu lcd ?

danny1231233, 2019-08-03:
now that the firmware is fully installed the lights go on but the extruder wont heat up and all the motors wont move. also the fans dont turn on

danny1231233, 2019-08-03:
yes i can navigate in menu lcd

luc-github, 2019-08-03:
go to settings then navigate to `Load Fail-Safe` then press Ok
then go to `Store -> EPPROM`

danny1231233, 2019-08-03:
ok, i did that but the motors fan and extruder still doesnt work.

luc-github, 2019-08-03:
What is the output of `M205` ?

danny1231233, 2019-08-03:
where can i find m205

luc-github, 2019-08-03:
Connect arduino serial terminal  to your printer - baud rate 230400 with NL/CR
then type M205 upper case M

danny1231233, 2019-08-03:
it says this

wawti

awti

awti

awti

awti

awti

awti

awti

awti

awti

ko0 

PE:R 001820 L naugga
eE
RP2:7  5324000B uardta
eE
RP0:1 21 5 1iDpsal yoMed
:E
RP0:1 11 9 1iLhg tnO
:E
RP0:1 21 7 0eKpeL githO :n

PE:R 011621 F limane teSsnroO :n

PE:R 011670 T poS neos rnO
:E
RP0:1 21 0 0oSnu dnO
:E
RP3:1 920 0.00F limane trpniet dm[
]E
RP2:1 520 P irtnrea tcvi es[
]E
RP2:7  9 0aM.xi antcvi eitem[ sm0,o=ff
]E
RP2:8  3 0tSpos etpprea tfrei antcviti ym[,s=0fo]f

PE:R 211120 P woreaseva tfre[ sm0,o=ff:]

PE:R 311060 0.00T me pxE tLP:A

PE:R 311460 0.00T me pxE tBA:S

PE:R 311860 0.00T me peB dLP:A

PE:R 311270 0.00T me peB dBA:S

PE:R 311970 0.00L ao deFdeR ta:e

PE:R 311380 0.00U lnao deFdeR ta:e

PE:R 311780 0.00U lnao/doLdaD siatcn:e

PE:R 3 3.00000X a-ix stspe sep rmm

PE:R 3 7.00000Y a-ix stspe sep rmm

PE:R 3110 0.00 0-Zxasis etspp rem 
mE
RP3:1  5.000 0-Xxasim xa .efdearet[ mms/
]E
RP3:1  9.000 0-Yxasim xa .efdearet[ mms/
]E
RP3:2  3.000 0-Zxasim xa .efdearet[ mms/
]E
RP3:2  7.000 0-Xxasih moni gefdearet[ mms/
]E
RP3:3  1.000 0-Yxasih moni gefdearet[ mms/
]E
RP3:3  5.000 0-Zxasih moni gefdearet[ mms/
]E
RP3:3  9.000 0aM.xj re km[/m]s

PE:R 3740 0.00M xa .-Zejkr[ mms/
]E
RP3:1 330 0.00X m nip so[ mm
]E
RP3:1 730 0.00Y m nip so[ mm
]E
RP3:1 140 0.00Z m nip so[ mm
]E
RP3:1 540 0.00X m xal netg hm[]m

PE:R 341 9.000 0 Yam xelgnht[ mm
]E
RP3:1 350 0.00Z m xal netg hm[]m

PE:R 3150 0.00X a-ix scaecelaritno[ mms/2^
]E
RP3:5  5.000 0-Yxasia ccleretaoi nm[/m^s]2

PE:R 3950 0.00Z a-ix scaecelaritno[ mms/2^
]E
RP3:6  3.000 0-Xxasit arev lcaecelaritno[ mms/2^
]E
RP3:6  7.000 0-Yxasit arev lcaecelaritno[ mms/2^
]E
RP3:7  1.000 0-Zxasit arev lcaecelaritno[ mms/2^
]E
RP3:1 20 4.040 0oCtani ghtcinkse sm[]m

PE:R 311821 000.00M naau-lrpbo e1X[ mm
]E
RP3:1 31 281.000 0aMunlap-orebY  1m[]m

PE:R 311631 000.00M naau-lrpbo e2X[ mm
]E
RP3:1 41 0010.00M naau-lrpbo e2Y[ mm
]E
RP3:1 41 4050.00M naau-lrpbo e3X[ mm
]E
RP3:1 41 8590.00M naau-lrpbo e3Y[ mm
]E
RP3:1 51 251.000 0aMunlap-orebX  4m[]m

PE:R 311659 .500 0aMunlap-orebY  4m[]m

PE:R 308 8.045 0-Zrpbo eehgith[ mm
]E
RP3:9 925 0.00M xa .-zrpbo e -eb didts .m[]m

PE:R 318 2.100 0-Zrpbo epsee dm[/m]s

PE:R 348 0030.00Z p-orebx y-s-epde[ mms/
]E
RP3:8 000 0.00Z p-orebo ffes t xm[]m

PE:R 308 4.000 0-Zrpbo efosftey [ mm
]E
RP3:8 613 .600 0-Zrpbo e1X[ mm
]E
RP3:8 02- .700 0-Zrpbo e1Y[ mm
]E
RP3:8 423 .600 0-Zrpbo e2X[ mm
]E
RP3:8 822 300.00Z p-orebY  2m[]m

PE:R 338 271.100 0-Zrpbo e3X[ mm
]E
RP3:8 632 300.00Z p-orebY  3m[]m

PE:R 301630 0.00Z p-orebb neidgnc roertcoi n Am[]m

PE:R 301040 0.00Z p-orebb neidgnc roertcoi n Bm[]m

PE:R 301440 0.00Z p-orebb neidgnc roertcoi n Cm[]m

PE:R 088 0 1uAotelev lcaitev( /1)0

PE:R 001 6 0eB deHtaM nagare[ -0]3

PE:R 001 7 0eB dIP Drdvi eam
xE
RP0:1 420 B deP DId irevm ni

PE:R 301 8.000 0eB dIP D-Pagni

PE:R 311 2.000 0eB dIP D-Iagni

PE:R 311 6.000 0eB dIP D-Dagni

PE:R 021 0 0eB dIP Dam xavul e0[2-55
]E
RP0:1 20 0 0nEbaelr teartcoi nocvnreisno[ /0]1

PE:R 399 2.300 0eRrtcaitnol netg hm[]m

PE:R 399 6310.00R teartcoi nelgnhte txured rwstihc[ mm
]E
RP3:1 00 0040.00R teartcoi npsee dm[/m]s

PE:R 301400 0.00R teartcoi n-ziltf[ mm
]E
RP3:1 00 8.000 0xErt axertsuoi nnou dn oerrtca tm[]m

PE:R 301210 0.00E txare txurisnoo  nnuods iwct herrtca tm[]m

PE:R 301612 .000 0eRrtcaitnou dn opsee
dE
RP3:2 000 0.00E tx.r 1tspe sep rmm

PE:R 302 4.000 0xErt1.m xa .efdearet[ mms/
]E
RP3:2 800 0.00E tx.r 1tsra tefdearet[ mms/
]E
RP3:2 210 0.00E tx.r 1caecelaritno[ mms/2^
]E
RP0:2 610 E tx.r 1ehtam nagare[ -0]3

PE:R 012 7 0xErt1.P DId irevm xa

PE:R 042 5 0xErt1.P DId irevm ni

PE:R 312 8.00000E tx.r 1IP D-Pagnid/ae-ditem

PE:R 322 2.00000E tx.r 1IP D-Iagni

PE:R 322 6.00000E tx.r 1IP D-Dagni

PE:R 032 0 0xErt1.P DIm xav laeu[ -052]5

PE:R 232 1 0xErt1.X o-ffes ts[etsp
]E
RP2:2 530 E tx.r 1-Yfosfte[ tspe]s

PE:R 292 0 0xErt1.Z o-ffes ts[etsp
]E
RP1:2 930 E tx.r 1etpm .tsbalizi eitem[ ]s

PE:R 152 0 0xErt1.t me.pf ror teartcoi nhwneh aeitgn[ ]C

PE:R 152 2 0xErt1.d siatcn eotr teartcw eh nehtani gm[]m

PE:R 052 452 5xErt1.e txured roclores epde[ -052]5

PE:R 342 6.000 0xErt1.a vdnaecL [ =0fo]f

PE:R 303 0.000 0xErt2.s etspp rem 
mE
RP3:3 400 0.00E tx.r 2am.xf eerdta em[/m]s

PE:R 303 8.000 0xErt2.s attrf eerdta em[/m]s

PE:R 313 2.000 0xErt2.a ccleretaoi nm[/m^s]2

PE:R 013 6 0xErt2.h ae tamaneg r0[3-
]E
RP0:3 710 E tx.r 2IP Drdvi eam
xE
RP0:3 540 E tx.r 2IP Drdvi eim
nE
RP3:3 810 0.00 0xErt2.P DIP g-ia/neddat-mi
eE
RP3:3 220 0.00 0xErt2.P DII g-ia
nE
RP3:3 620 0.00 0xErt2.P DID g-ia
nE
RP0:3 030 E tx.r 2IP Dam xavul e0[2-55
]E
RP2:3 130 E tx.r 2-Xfosfte[ tspe]s

PE:R 233 5 0xErt2.Y o-ffes ts[etsp
]E
RP2:3 090 E tx.r 2-Zfosfte[ tspe]s

PE:R 133 9 0xErt2.t me.ps atibilezt mi es[
]E
RP1:3 050 E tx.r 2etpm .of rerrtcaitnow eh nehtani gC[
]E
RP1:3 250 E tx.r 2idtsnaect  oerrtca thwneh aeitgn[ mm
]E
RP0:3 452 55E tx.r 2xertdurec ooel rpsee d0[2-55
]E
RP3:3 640 0.00E tx.r 2daavcn e L0[o=ff
]w
ia
tw
ia
tw
ia
tw

luc-github, 2019-08-03:
your communication has issue, the output is terrible and your EEPROM is not correct

Try ` M502` then `M500` again

danny1231233, 2019-08-03:
ia
tw
ia
to
 k
0I
fn:ouAotelevilgnd sibael
dZ
j re kaw sot ool,ws teitgnt  o.043

eSeltcxErtdure0:

lFwouMtlpiyl1:00

nIofC:noifugaritnor setet  oedaflust
.w
ia
tw
ia
tw
ia
tw
ia
tw
ia
to
 k
0I
fn:ooCfngirutaoi ntsrodet  oEERPMO
.w
ia
tw

luc-github, 2019-08-03:
when you go to printer menu what is the `Baudrate` in settings ?
when you go to position menu and select `Home all`, what happen ?

danny1231233, 2019-08-03:
230400

now it homes it.

luc-github, 2019-08-03:
so it should heat too

luc-github, 2019-08-03:
it is heating ?

danny1231233, 2019-08-03:
yes it is working now. Thank You so much for your help

luc-github, 2019-08-03:
Good

closed 2019-08-03 by luc-github