luc-github/Repetier-Firmware-4-Davinci #352: [BUG]

Handt21, 2019-10-22:
I've followed every I could do and keeps saying uns device not recognized
[image]

luc-github, 2019-10-22:
printer is working ?

Handt21, 2019-10-22:
No it's not working. I still have black bars on lcd

luc-github, 2019-10-22:
so you followed FAQ about black bars ?

Handt21, 2019-10-22:
Yes I've tried to fallow them but it's a little  confusing. Is anyone willing to help maybe teamviewer

luc-github, 2019-10-22:
what is confusing ?
if you are not able to compile, why you do not flash the binary present in release ?
https://github.com/luc-github/Repetier-Firmware-4-Davinci/issues/109#issuecomment-373344462

Handt21, 2019-10-22:
How do you do that.

luc-github, 2019-10-22:
Did you click on link I gave you ?

Handt21, 2019-10-22:
I will try it all over after work today

Handt21, 2019-10-22:
So I believe I've done everything correctly but it stops at 79%

luc-github, 2019-10-23:
https://github.com/luc-github/Repetier-Firmware-4-Davinci/issues/185

Handt21, 2019-10-23:
I got it. Sorry I didnt message back right away. Thank you for the help.

luc-github, 2019-10-23:
good

closed 2019-10-23 by luc-github