luc-github/Repetier-Firmware-4-Davinci #353: "Error compiling for board Arduino Due (Native USB port)."

cadmadness, 2019-10-24:
Followed instructions as best as I could (first time using Arduino) and as I go to verify and/or upload the .ino it gives an error message in the subject line. Any ideas where I went wrong? I am using a Mac to run Arduino 1.8.0 . Thanks in advance!

Arduino: 1.8.0 (Mac OS X), Board: "Arduino Due (Native USB Port)"

/Users/matt/Desktop/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Users/matt/Desktop/Arduino.app/Contents/Java/hardware -hardware /Users/matt/Library/Arduino15/packages -tools /Users/matt/Desktop/Arduino.app/Contents/Java/tools-builder -tools /Users/matt/Desktop/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/matt/Library/Arduino15/packages -built-in-libraries /Users/matt/Desktop/Arduino.app/Contents/Java/libraries -libraries /Users/matt/Documents/Arduino/libraries -fqbn=arduino:sam:arduino_due_x -ide-version=10800 -build-path /var/folders/tb/bnps9_y568qglh5px463pmhr0000gn/T/arduino_build_302843 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/Users/matt/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -prefs=runtime.tools.bossac.path=/Users/matt/Library/Arduino15/packages/arduino/tools/bossac/1.6.1-arduino -verbose /Users/matt/Desktop/Repetier/Repetier.ino
/Users/matt/Desktop/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Users/matt/Desktop/Arduino.app/Contents/Java/hardware -hardware /Users/matt/Library/Arduino15/packages -tools /Users/matt/Desktop/Arduino.app/Contents/Java/tools-builder -tools /Users/matt/Desktop/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/matt/Library/Arduino15/packages -built-in-libraries /Users/matt/Desktop/Arduino.app/Contents/Java/libraries -libraries /Users/matt/Documents/Arduino/libraries -fqbn=arduino:sam:arduino_due_x -ide-version=10800 -build-path /var/folders/tb/bnps9_y568qglh5px463pmhr0000gn/T/arduino_build_302843 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/Users/matt/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -prefs=runtime.tools.bossac.path=/Users/matt/Library/Arduino15/packages/arduino/tools/bossac/1.6.1-arduino -verbose /Users/matt/Desktop/Repetier/Repetier.ino
Using board 'arduino_due_x' from platform in folder: /Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8
Using core 'arduino' from platform in folder: /Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8
Detecting libraries used...
"/Users/matt/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf  -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10800 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Due"' "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/libsam" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/CMSIS/CMSIS/Include/" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/CMSIS/Device/ATMEL/" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/cores/arduino" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/variants/arduino_due_x" "/var/folders/tb/bnps9_y568qglh5px463pmhr0000gn/T/arduino_build_302843/sketch/Repetier.ino.cpp" -o "/dev/null"
"/Users/matt/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf  -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10800 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Due"' "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/libsam" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/CMSIS/CMSIS/Include/" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/CMSIS/Device/ATMEL/" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/cores/arduino" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/variants/arduino_due_x" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/libraries/SPI/src" "/var/folders/tb/bnps9_y568qglh5px463pmhr0000gn/T/arduino_build_302843/sketch/Repetier.ino.cpp" -o "/dev/null"
"/Users/matt/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf  -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10800 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Due"' "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/libsam" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/CMSIS/CMSIS/Include/" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/CMSIS/Device/ATMEL/" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/cores/arduino" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/variants/arduino_due_x" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/libraries/SPI/src" "/var/folders/tb/bnps9_y568qglh5px463pmhr0000gn/T/arduino_build_302843/sketch/BedLeveling.cpp" -o "/dev/null"
"/Users/matt/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf  -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10800 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Due"' "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/libsam" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/CMSIS/CMSIS/Include/" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/CMSIS/Device/ATMEL/" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/cores/arduino" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/variants/arduino_due_x" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/libraries/SPI/src" "/var/folders/tb/bnps9_y568qglh5px463pmhr0000gn/T/arduino_build_302843/sketch/Commands.cpp" -o "/dev/null"
"/Users/matt/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf  -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10800 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Due"' "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/libsam" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/CMSIS/CMSIS/Include/" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/CMSIS/Device/ATMEL/" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/cores/arduino" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/variants/arduino_due_x" "-I/Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/libraries/SPI/src" "/var/folders/tb/bnps9_y568qglh5px463pmhr0000gn/T/arduino_build_302843/sketch/Communication.cpp" -o "/dev/null"
[... cut ...]
"/Users/matt/Desktop/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/var/folders/tb/bnps9_y568qglh5px463pmhr0000gn/T/arduino_build_302843/preproc/ctags_target_for_gcc_minus_e.cpp"
Using library SPI at version 1.0 in folder: /Users/matt/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/libraries/SPI 
fork/exec /Users/matt/Desktop/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags: bad CPU type in executable
Error compiling for board Arduino Due (Native USB Port).

luc-github, 2019-10-24:
what is your OS version ?

cadmadness, 2019-10-24:
I am on macOS Catalina. I had a feeling this might have been a reason (maybe something with being 64bit only apps?) so borrowed another computer on a much older OS and was successful in flashing the printer. Thank you for your response!

luc-github, 2019-10-24:
yes I see several people having issue with catalina and arduino. hope they will fix this soon

luc-github, 2019-10-24:
Happy you managed it
Happy printing ^_^

closed 2019-10-24 by luc-github

cadmadness, 2019-10-24:
Thank you very much for putting this all together it really made it a fun challenge for me to get it flashed. Now I have to figure out how to get the bed set properly, I attempted to print something and the nozzle right on the surface. 

Thanks again!

cadmadness, 2019-10-25:
Sorry to bug you again but I cannot seem to find an answer online. Every time I try to print something the extruder is so close to the base it is practically scratching it. What setting and how might I find it on Repetier to make sure this doesn’t take place?

luc-github, 2019-10-25:
as stated in readme 
```
11. When update is complete you must calibrate your bed height!Use manual bed leveling in menu
```
you need to redo you bed leveling with a paper sheet and screws , if your extruder scratch bed it means your bed is too close

Timboslice8, 2023-10-25:
I need help! iv been at this for days

Arduino: 1.8.0 (Windows 10), Board: "Arduino Due (Native USB Port)"

Board arduino_due_x (platform sam, package arduino) is unknown

Error compiling for board Arduino Due (Native USB Port).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.