luc-github/Repetier-Firmware-4-Davinci #354: Can't Heat the Extruder on a AIO

Wob76, 2019-11-05:
Hi,

I have flashed my Da Vinco AIO using the master branch (0.92.10). Motors are working as expected, Temp Sensors seem good (reading room temp), Bed heats fine but the extruder just won't heat up. I have issued M502 and M500 commanded, I also tried resetting from the LCD Menu. I get and error about the heater seems to have decoupled error due to it not moving from room temp, it also remains cool to touch. It was all working fine with the stock firmware.

**To Reproduce**
Steps to reproduce the behavior:
Issue is triggered by manually enabling the extruder in the LCD menu, via Repetier-Host, GCode or trying to do a bed level.

**PC (please complete the following information):**
 - OS: Windows 10
 - Arduino Version - 1.8.0
 - Due Module Version - 1.6.8

**Printer
AiO

**Firmware (please complete the following information):**
- Master Branch
- 0.9.10
- configuration.h define DAVINCI 4 and MODEL  1
- Did you modified the code ? Nope

The only time I saw any issue heating the extruder with stock firmware is when I once tried to unload AFTER removing the printer cartridge, seems it has a fail safe, re-installing the cartridge and started working. I have tried with and without the cartridge and no change, but I assume the new firmware will not be checking the cartridge anyway.

Any tests to complete will be appreciated.

Thanks,
Beau

luc-github, 2019-11-05:
* if printer is behaving normally but this issu,  this it is not FW issue, unless XYZ changed the pin out, which I doubt
* if bed is heating, it is not a power issue, 12V is working,

 It is likely a connector issue  or any wire continuity issue. check the heater connector  and continuity.
It is common issue on Davinci
 
you can also check this thread http://www.soliforum.com/topic/6324/xyz-printer-extruder-not-heating-up-properly/

Wob76, 2019-11-05:
Hi,

Thanks for the response, I re flashed the unit using the release bin file and bossac to see if that was a cause, it was still the same, checked for extruder voltage on the board and it was indeed going to 12V when the extruder was turned on. 

I then unpluged the exturder expecting that plug on top to be the issue as it is widely reported, Heater element passed the continuity test and I was also getting 12V on the power side, reconnected it and it is all working as expected, maybe it bumped loose while I was moving the printer about during the reset.

So all good, thanks again for your help.

closed 2019-11-05 by Wob76