luc-github/Repetier-Firmware-4-Davinci #355: Calibration of Horus with AiO

brimstone326, 2020-01-03:
Hello,

I have been following this for some time and I recently acquired an AiO printer which I converted to Repetier.  I would like to find a way to get the scanner working and am happy to help testing.  I just cannot seem to get the Horus software to calibrate.  I tried with some difficulty to follow all the French conversations with zeus10.  I know you closed the ticket some time ago, but if you are willing to pursue this, I would be happy to try to get it running.  I can connect to Horus and see the target on the bed.  Horus appears to function in its attempts to rotate the table and calibrate, but the interior of the machine seems to be too dark no matter what settings I use.  I never see the multi-colored lines that identify the squares.

As a backup, I don't know how difficult it would be to create a set-up for it, but ReconstructMe software works with an Xbox Kinect.  Could it be possibly used to connect to the cameras and scan? 

http://reconstructme.net/

Thanks for all your work on this.  Even without the scanner, the printer is free of all the XYZ issues.

luc-github, 2020-01-03:
hi, I do not have AiO so I have only worked on motor part
for the scan part and horus support there is no progress for a while
the AiO cameras setup is different than a kinect so I cannot answer about difficulty
any help is welcome if you have an AIO

brimstone326, 2020-01-04:
I made a calibration target that fits the AiO's table.  I tried calibrating today and managed to get the rainbow lines to show up for a second, but I moved the slider and never got it to come back.  Did you ever make the Ciclops scanner?  And if so, did you have issues getting it to focus well enough to capture the squares?  I have seen a few videos on calibrating the software, but they are using the older version software and when they show the calibration, it's already dialed in.... so there's no mention of how many hours they spent getting it to read properly.  I'm tempted to try the older version, but I don't know if it will be any easier to dial in.

[https://www.thingiverse.com/thing:4084516]

luc-github, 2020-01-04:
I tried to build one some years ago but never finished it, horus developper stopped support at this time and I was too busy with others project to put more time on this project

brimstone326, 2020-01-21:
Hello,  I think I'm going to close this for the time being.  I have been able to get some lines for calibration, but never stable enough to complete the process.  The focal point of the AiO cameras does not seem to be the center of the platform.  I took apart one of the assemblies to see if I could find any internal adjustment or identify the camera module to find out where the focus should be.  I had no luck there.  My goal now is to print a custom Cyclops-style housing utilizing one camera and the two lasers.  Hopefully I can make it adjustable enough to get through calibration.

closed 2020-01-21 by brimstone326