luc-github/Repetier-Firmware-4-Davinci #356: Error:PID Autotune failed! timeout

jmragin, 2020-01-18:
I tried to run the PID autotune calibration with:
M303 E0 S200 C8

I got the following:
Info:PID Autotune start
< 9:12:56 PM:  bias: 99 d: 99 min: 196.50 max: 205.93
< 9:13:21 PM:  bias: 75 d: 75 min: 196.84 max: 205.74
< 9:23:13 PM: Error:PID Autotune failed! timeout

Any suggestions?

luc-github, 2020-01-18:
may be with correct parameters  :
[
`M303 P<extruder/bed> S<printTemerature> X0 R<Repetitions>- Autodetect pid values. Use P<NUM_EXTRUDER> for heated bed. X0 saves result in EEPROM. R is number of cycles.`
](https://github.com/repetier/Repetier-Firmware/blob/master/src/ArduinoDUE/Repetier/Repetier.ino#L145)

jmragin, 2020-01-19:
Nope, still failed. Failed after 10 minutes.

I tried this this time:
M303 P0 S200 R8 C0

luc-github, 2020-01-19:
did you set PID for extruder heat manager? by default it is 3 = Dead-time control

luc-github, 2020-01-19:
when Heat manager is is set to 1 andusing `M303 P0 S200`  after few minutes I got:

```
[...]
T:200.00 / 0 B:27.52 / 0 B@:0 @:86 T0:200.00 / 0 @0:86 T1:25.14 / 0 @1:0
busy:heating
 bias: 170 d: 84 min: 198.85 max: 201.41
 Ku: 41.81 Tu: 23.94
 Classic PID
 Kp: 25.09
 Ki: 2.10
 Kd: 75.08
Info:PID Autotune finished ! Place the Kp, Ki and Kd constants in the Configuration.h or EEPROM
T:199.89 / 0 B:27.43 / 0 B@:0 @:0 T0:199.89 / 0 @0:0 T1:25.41 / 0 @1:0
[...]
```
So it is ok

jmragin, 2020-01-20:
I re-flashed my DaVinci AiO with the following setting:
#define EXT0_HEAT_MANAGER 1

I am still getting the same results.  Extruder temp goes up past 200 three times and then stay around 183-184.

jmragin, 2020-01-20:
Oh and how do I get similar out put? Is it some sort of debugging option?

luc-github, 2020-01-20:
once flashed value is in EEPROM, you need to change in EEPROM
output is on serial

jmragin, 2020-01-20:
I changed the value in EEPRO and I am still having the same issue.  Printer heats up past 200 three times and then goes down to about 183-184 and stays there until PID Autotune failed! timeout.

luc-github, 2020-01-20:
well I do not know then, I cannot reproduce so it may be some settings, but as mentioned before this is pure repetier code, not related to Davinci, so not sure I can help on this sorry.

closed 2020-02-13 by stale[bot]