luc-github/Repetier-Firmware-4-Davinci #357: [Question]Migration Repetier : The fool who missed the flashing of his printer

Tchicken95, 2020-03-29:
Good morning all,

I made bad manipulations with my Da Vinci 2.0a Duo second generation ;(

I used the last Arduino 1.8.12

Is it possible to catch the following errors :

the first time, I flashed with the following configuration :

[code]// ################## EDIT THESE SETTINGS MANUALLY ################

#define DAVINCI 1// "0" if not DAVINCI, "1" For DAVINCI 1.0, "2" For DAVINCI 2.0 with 1 FAN, "3" For DAVINCI 2.0 with 2 FAN, 4 for AiO (WITH NO SCANNER SUPPORT)
#define MODEL  0//"0" for first generation (jumper JP1 to reset ) , "1" for new generation   (jumper J37 to reset)
#define REPURPOSE_FAN_TO_COOL_EXTRUSIONS 0 //Setting this to 1 will repurpose the main Extruder cooling fan to be controlled VIA M106/M107
                                           //Warning: for DaVinci 1.0 need to add a permanent fan with power supply to cool extruder

// ################ END MANUAL SETTINGS ##########################[/code]

the second time, I flashed with the following configuration :

[code]// ################## EDIT THESE SETTINGS MANUALLY ################

#define DAVINCI 3// "0" if not DAVINCI, "1" For DAVINCI 1.0, "2" For DAVINCI 2.0 with 1 FAN, "3" For DAVINCI 2.0 with 2 FAN, 4 for AiO (WITH NO SCANNER SUPPORT)
#define MODEL  0//"0" for first generation (jumper JP1 to reset ) , "1" for new generation   (jumper J37 to reset)
#define REPURPOSE_FAN_TO_COOL_EXTRUSIONS 0 //Setting this to 1 will repurpose the main Extruder cooling fan to be controlled VIA M106/M107
                                           //Warning: for DaVinci 1.0 need to add a permanent fan with power supply to cool extruder

// ################ END MANUAL SETTINGS ##########################[/code]

the third time, I flashed with the right configuration :

[code]// ################## EDIT THESE SETTINGS MANUALLY ################

#define DAVINCI 3// "0" if not DAVINCI, "1" For DAVINCI 1.0, "2" For DAVINCI 2.0 with 1 FAN, "3" For DAVINCI 2.0 with 2 FAN, 4 for AiO (WITH NO SCANNER SUPPORT)
#define MODEL  1//"0" for first generation (jumper JP1 to reset ) , "1" for new generation   (jumper J37 to reset)
#define REPURPOSE_FAN_TO_COOL_EXTRUSIONS 0 //Setting this to 1 will repurpose the main Extruder cooling fan to be controlled VIA M106/M107
                                           //Warning: for DaVinci 1.0 need to add a permanent fan with power supply to cool extruder

// ################ END MANUAL SETTINGS ##########################[/code]

after all this nonsense, windows loops on printer recognition, but never recognize it !

The printer screen has two nice black bars ;(

I tried several times to flash with the right configuration, but I still have the same result.

Do you think I shot my card ?

What can I do ?

Thank you for your help.

Tchicken.

luc-github, 2020-03-29:
Did you read :
1 - Readme
2 - pinned issue about black bar
3 - FAQ

Tchicken95, 2020-03-29:
Hello, 

 

I'm sorry, my English is not very good, I live in France, I will look at all that you have indicated to me.

Thank you for your help. 

 

Sincerely, Michel Poulet.

luc-github, 2020-03-29:
google translate est ton ami ^_^

Tchicken95, 2020-03-29:
Thank you for your help, I think I followed your procedure correctly, description below in French.
Can you tell me where I could make a mistake, the printer is not recognized by windows and I always have the two bars on the screen.

Rendez-vous à l’adresse :
https://github.com/luc-github/Repetier-Firmware-4-Davinci
Cliquez sur « Download ZIP »
Ensuite, rendez-vous à l’adresse :
https://www.arduino.cc/en/Main/OldSoftwareReleases#previous
Puis téléchargez la version 1.8.0 qui se trouve plus bas dans la page
Maintenant, préparez votre imprimante :
-	Branchez le câble USB à votre ordinateur
-	Eteignez l’imprimante
-	Enlevez les 2 vis du cache arrière
-	Munissez-vous d’un petit fil de fer que vous pliez en « U », il va servir à créer le pont de réinitialisation du Firmware.
-	Repérez sur la carte mère le jumper nommé J1 sur les anciennes versions ou J37 sur les versions suivantes
-	Créez votre pont en mettant en contact votre « U » sur les 2 petits contacts du jumper.
-	Maintenez votre pont et allumez l’imprimante, au bout de 5 secondes, votre carte mère est flashée !
Maintenant, installez Arduino 1.8.0 sur votre ordinateur.
Installer les pilotes non connus par Windows.
Dézippez le fichier :
Repetier-Firmware-4-Davinci-master.zip
Lancez l’application Arduino.
Puis faites « Fichier » et « Ouvrir »
Sélectionnez le fichier « Repetier.ino »
Puis cliquez sur « Ouvrir »
Dans la nouvelle fenêtre Arduino :
Cliquer sur « Outils », allez à « Type de carte » puis cliquez sur « Gestionnaire de cartes »
Faites défiler et sélectionnez :
« arduino SAM Board (32-bitsARM Cortex-M3) by Arduino version 1.6.12 INSTALLED »
Sélectionnez la version 1.6.8 et cliquez sur « Installer »
Fermez le gestionnaire de cartes.
Cliquer sur « Outils », allez à « Type de carte » puis sélectionnez « Arduino DUE (Native USB Port) »
Cliquer sur « Outils », allez à « Port : » puis sélectionnez « COM5 »
Ensuite, cliquez sur « Fichier » puis « Préférences »
Dans la nouvelle fenêtre, cliquez sur « Parcourir »
Puis sélectionnez le répertoire suivant dans l’arborescence du package « Repetier-Firmware-4-Davinci-master » :
Repetier-Firmware-4-Davinci-master\src\ArduinoDUE\AdditionalArduinoFiles\Arduino - 1.8.0 -Due 1.6.8\Arduino15\packages\arduino\hardware\sam\1.6.8
Cliquez sur l’onglet « Configuration.h » et modifiez les ligne suivantes :
#define DAVINCI 3	Pour le modèle 2.0a Duo
#define MODEL  1	"0" si vous avez un jumper « JP1 », "1" pour un jumper « J37 »
Ctrl + s pour enregistrer les modifications.
Cliquez sur le bouton de compilation en forme de « V » :
Puis cliquez sur le bouton « Téléverser », en forme de flèche droite

Thanks for your help.
Tchicken.

luc-github, 2020-03-29:
Je ne sais pas d'ou ca sort mais il manque le principal: l'installation de variants.cpp et USBCore.cpp
le readme est clair et se traduit facilement avec google translate https://translate.google.com/#view=home&op=translate&sl=en&tl=fr

https://github.com/luc-github/Repetier-Firmware-4-Davinci#installation
la FAQ:
https://github.com/luc-github/Repetier-Firmware-4-Davinci/issues/172
https://github.com/luc-github/Repetier-Firmware-4-Davinci/issues/114

Tchicken95, 2020-03-29:
Merci pour ces informations et ces promptes réponses.
Tous c'est bien passé, du coup j'ai pris le temps de faire un tutoriel précis et en français pour l'installation des Da Vinci 2.0a Duo sous Windows. Il est disponible sur le forum suivant : 
https://www.lesimprimantes3d.fr/forum/topic/29302-quel-logiciel-dimpression-choisir/?tab=comments#comment-339583
Bravo à toute l'équipe pour ce formidable boulot.
Je vais maintenant me consacrer à la création d'objets pour aider le service hospitalier de ma femme qui en a bien besoin en ce moment.

closed 2020-03-29 by Tchicken95