luc-github/Repetier-Firmware-4-Davinci #358: Stepper motor knocking

3-ma, 2020-04-06:
What is your question ?
Hi Luc! Congrats on the work about this printer. I got one that was broken and managed to fix it, then put Repetier following your guide.

Everything works fine, exept on big objects: sometimes, but kinda in the same position every time, the stepper starts knocking and the filament won't melt, so the printer will go on, but the extruder does nothing. So there's no filament being extrudent, but the axis moves and the print goes on. I have to push the filamente with a bit of force and usually it starts to get pulled. But again, in that area of the print, the knocking sound starts and I have to push the filament.
It results in canceled prints, since many parts get skipped by the filament.

I read it might be not enough current to the stepper motor (I already cleaned the nozzle, tried a higher temperature, cleaned the gears that pull the filament). Since the problem appears usually in the same area, with fast movements or supports creation, I think it could be caused by low current.
Did you ever encountered this issue?

luc-github, 2020-04-06:
>the filament won't melt,
this should not be linked to steeper or low current driver -  you may have another issue

instead of increase temperature try to lower the print speed, 50mm/sec is a max for davinci

3-ma, 2020-04-06:
It doesn't melt because there's no melted filament being extruded, and it could depend on several issues. I notice that when it's knocking, the filament don't get pulled and sometimes it get pushed back, then it still knock a bit and then it god pulled correctly.

60mm max?? On the specs page of the producer it says 120mm/s! I was printing at 60mm/s by Cura's default, but I was going to try 50mm/s before you're reply.
I'll start a print at 50mm/s and see what happens.

Thank you for your support.

3-ma, 2020-04-06:
Can confirm. Even at 50mm/s it knocks. Is there a way to push current from firmware? Or do I need to do it from the mainboard? It'll be a pain in the..... since there's no way to manage it from the board. Justo to try something.

luc-github, 2020-04-06:
no - there is no current control from FW  - the drivers are standalone's one and unfortunatly no potentiometers
 
hmm I never got such issue - but if you said :
>Since the problem appears usually in the same area, with fast movements 

It is really looks wiring issue

3-ma, 2020-04-06:
Do you mean the filament is trash? Or that the stepper can't feed the extruder properly? The issue happens on low X and Y, so pretty near the spool. I launched another print with the spool placed in another way, let's try. Thanks.

3-ma, 2020-04-08:
Hi Luc, I managed to set the spool on a stable position, but the issue still happens. This evening I'm going to remove the extruder and let it soak in acetone. Looks like the motor tries to spin but the filament stays still. So, the motor's gear scratches the filaments. I'll update the issue tomorrow.

luc-github, 2020-04-08:
what filament are you using with stock extruder ?

3-ma, 2020-04-08:
XZYPrinting PLA. The issue appears both at 190°C and 200°C.
The extruder is far from clogged, I pushed a thin metal wire through it quite often in these days. I'll try other ways to clean it.
Yesterday I swapped motors, gears and springs between the extruders, but today the issue is still there. The only part I didn't swapped are the extruders themselves. I'll clean it, try it, if still knocks, I'll switch extruders. If still knocks, I'll trow the printer from the window.

luc-github, 2020-04-08:
Hmm sorry you cannot use normal PLA with stock extruder - this the problem, only abs is working well with stock extruder
the only pla working with stock extruder,... is the XYZ one, because it is not real PLA...

if you want to use to use PLA, you better change extruder

One temporary solution because I tested long time ago: 
1 -  the filament must be out of the roll so extruder do no need pulling the filament strongly
2 - put some cooking oil on a textile and make your filament going thrue this textile to lubricate it slighly
this is ugly, but it is the only way I found to use stock extruder and non xyz PLA  filament

luc-github, 2020-04-08:
Also did you tried another spool to see if issue come from your filament, (dust, moisture, etc) ?

Another test : does it happen when your extruder is far from bed ?
 try print some on the air 
if not the issue may be due to that you are too close from bed or over extrusing,  and then your extruder clog because it cannot extrude properly

3-ma, 2020-04-09:
> One temporary solution because I tested long time ago:
> 1 - the filament must be out of the roll so extruder do no need pulling the filament strongly

It's outside the cartridge but still as spool. I placed it to minimize the friction, really, the stepper doesn't need to pull much the filament.

> Try to lub the filament it should still help

Ok, I'll lube it all for ya!

> does it happen when your extruder is far from bed ?

It just happened around layer 1 or 2, the extruder was really close to the bed since they were the first layers. But it happened on other higher layer on other prints.. Like 10mm from the bed.
I'll look up the over-extruding issue, since during the same print the issue happens often in the same area.

> may be due to that you are too close from bed or over extrusing

the last time it happened, I noticed the bed wasn't even (like 0.5mm difference from the opposing sides, and the problem appeared where the extruder was closer to the bed. The extruded filament was splattered and larger than the usual, so the nozzle was spreading it.

Today I'll test things out. Thank you again!

3-ma, 2020-04-30:
Hey Luc, mark this as solved. Using olive oil fixed the jamming, the issue never happened a single time once I lubed it.

3-ma, 2020-04-30:
And of course, thank you very much for the insight.

closed 2020-04-30 by 3-ma