luc-github/Repetier-Firmware-4-Davinci #359: Can't connect on USB

TexPonch, 2020-04-09:
System: Mac High Sierra 10.13.16 (early 2011) 
I have followed all the steps copied over the variants/USBCore
/Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/variants/arduino_due_x
and 
/Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/cores/arduino/USB

Running Arduino 1.8.13 / DUE 1.6.8 

I can't see that my printer is plugged in (yes it is jumpered/black bar, turned on and plugged in)

When I go to upload I get this error:
Arduino: 1.8.13 Hourly Build 2020/03/25 04:12 (Mac OS X), Board: "Arduino Due (Native USB Port)"

/Applications/Arduino 3.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino 3.app/Contents/Java/hardware -hardware /Users/alejandrogarcia/Library/Arduino15/packages -tools /Applications/Arduino 3.app/Contents/Java/tools-builder -tools /Applications/Arduino 3.app/Contents/Java/hardware/tools/avr -tools /Users/alejandrogarcia/Library/Arduino15/packages -built-in-libraries /Applications/Arduino 3.app/Contents/Java/libraries -libraries /Users/alejandrogarcia/Documents/Arduino/libraries -fqbn=arduino:sam:arduino_due_x -vid-pid=0000_0000 -ide-version=10813 -build-path /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_31562 -warnings=none -build-cache /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_cache_353903 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=/Users/alejandrogarcia/Library/Arduino15/packages/arduino/tools/bossac/1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=/Users/alejandrogarcia/Library/Arduino15/packages/arduino/tools/bossac/1.6.1-arduino -prefs=runtime.tools.arm-none-eabi-gcc.path=/Users/alejandrogarcia/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=/Users/alejandrogarcia/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -verbose /Users/alejandrogarcia/Desktop/Restting Printer/Repetier-Firmware-4-Davinci/src/ArduinoDUE/Repetier/Repetier.ino
/Applications/Arduino 3.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino 3.app/Contents/Java/hardware -hardware /Users/alejandrogarcia/Library/Arduino15/packages -tools /Applications/Arduino 3.app/Contents/Java/tools-builder -tools /Applications/Arduino 3.app/Contents/Java/hardware/tools/avr -tools /Users/alejandrogarcia/Library/Arduino15/packages -built-in-libraries /Applications/Arduino 3.app/Contents/Java/libraries -libraries /Users/alejandrogarcia/Documents/Arduino/libraries -fqbn=arduino:sam:arduino_due_x -vid-pid=0000_0000 -ide-version=10813 -build-path /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_31562 -warnings=none -build-cache /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_cache_353903 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=/Users/alejandrogarcia/Library/Arduino15/packages/arduino/tools/bossac/1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=/Users/alejandrogarcia/Library/Arduino15/packages/arduino/tools/bossac/1.6.1-arduino -prefs=runtime.tools.arm-none-eabi-gcc.path=/Users/alejandrogarcia/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=/Users/alejandrogarcia/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -verbose /Users/alejandrogarcia/Desktop/Restting Printer/Repetier-Firmware-4-Davinci/src/ArduinoDUE/Repetier/Repetier.ino
Using board 'arduino_due_x' from platform in folder: /Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8
Using core 'arduino' from platform in folder: /Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8
Detecting libraries used...
/Users/alejandrogarcia/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10813 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/libsam -I/Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/CMSIS/CMSIS/Include/ -I/Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/CMSIS/Device/ATMEL/ -I/Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/cores/arduino -I/Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/variants/arduino_due_x /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_31562/sketch/Repetier.ino.cpp -o /dev/null
Alternatives for SPI.h: [SPI@1.0]
ResolveLibrary(SPI.h)
  -> candidates: [SPI@1.0]
/Users/alejandrogarcia/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10813 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/libsam -I/Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/CMSIS/CMSIS/Include/ -I/Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/CMSIS/Device/ATMEL/ -I/Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/cores/arduino -I/Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/variants/arduino_due_x -I/Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/libraries/SPI/src /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_31562/sketch/Repetier.ino.cpp -o /dev/null
Using cached library dependencies for file: /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_31562/sketch/BedLeveling.cpp
Using cached library dependencies for file: /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_31562/sketch/Commands.cpp
Using cached library dependencies for file: /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_31562/sketch/Communication.cpp
Using cached library dependencies for file: /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_31562/sketch/Drivers.cpp
Using cached library dependencies for file: /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_31562/sketch/Eeprom.cpp
Using cached library dependencies for file: /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_31562/sketch/Extruder.cpp
Using cached library dependencies for file: /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_31562/sketch/HAL.cpp
Using cached library dependencies for file: /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_31562/sketch/Printer.cpp
Using cached library dependencies for file: /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_31562/sketch/SDCard.cpp
Using cached library dependencies for file: /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_31562/sketch/SdFat.cpp
Using cached library dependencies for file: /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_31562/sketch/gcode.cpp
Using cached library dependencies for file: /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_31562/sketch/motion.cpp
Using cached library dependencies for file: /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_31562/sketch/ui.cpp
Using cached library dependencies for file: /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_31562/sketch/uilang.cpp
Using cached library dependencies for file: /Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/libraries/SPI/src/SPI.cpp
Generating function prototypes...
[... cut ...]
PORTS {/dev/cu.Bluetooth-Incoming-Port, /dev/tty.Bluetooth-Incoming-Port, } / {/dev/cu.Bluetooth-Incoming-Port, /dev/tty.Bluetooth-Incoming-Port, } => {}
Uploading using selected port: /dev/cu.Bluetooth-Incoming-Port
/Users/alejandrogarcia/Library/Arduino15/packages/arduino/tools/bossac/1.6.1-arduino/bossac -i -d --port=cu.Bluetooth-Incoming-Port -U true -e -w -v -b /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_31562/Repetier.ino.bin -R 
Set binary mode
No device found on cu.Bluetooth-Incoming-Port
Send auto-baud
An error occurred while uploading the sketch
Set binary mode

Thanks in adv!

luc-github, 2020-04-09:
This issue is not linked to FW, but it is arduino ide flashing issue

check you can access to the port with your user's rights

of course be sure it is the right port the board is connected

I do not flash under osx so I cannot help so much - I add label to ask help if someone did it

TexPonch, 2020-04-09:
Ok so I have the "arduino portion" all correct?  I can see the printer plugged in to my mac but since I jumpered it the Mac sees the printer connection like a phone.
[image]

TexPonch, 2020-04-09:
The Mac is seeing the printer but as a modem/phone. 

[image]

luc-github, 2020-04-09:
using Arduino: 1.8.13 Hourly Build 2020/03/25 04:12 (Mac OS X) is bad idea - it is dev version so it may have bug 
use stable one - it was working in past and code did not changed

TexPonch, 2020-04-09:
ok deleted and ran 1.8.12

Looks like same error:

Arduino: 1.8.12 (Mac OS X), Board: "Arduino Due (Native USB Port)"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/alejandrogarcia/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/alejandrogarcia/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/alejandrogarcia/Documents/Arduino/libraries -fqbn=arduino:sam:arduino_due_x -vid-pid=0000_0000 -ide-version=10812 -build-path /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_401390 -warnings=none -build-cache /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_cache_728184 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/Users/alejandrogarcia/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=/Users/alejandrogarcia/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -prefs=runtime.tools.bossac.path=/Users/alejandrogarcia/Library/Arduino15/packages/arduino/tools/bossac/1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=/Users/alejandrogarcia/Library/Arduino15/packages/arduino/tools/bossac/1.6.1-arduino -verbose /Users/alejandrogarcia/Desktop/Restting Printer/Repetier-Firmware-4-Davinci/src/ArduinoDUE/Repetier/Repetier.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/alejandrogarcia/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/alejandrogarcia/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/alejandrogarcia/Documents/Arduino/libraries -fqbn=arduino:sam:arduino_due_x -vid-pid=0000_0000 -ide-version=10812 -build-path /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_401390 -warnings=none -build-cache /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_cache_728184 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/Users/alejandrogarcia/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=/Users/alejandrogarcia/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -prefs=runtime.tools.bossac.path=/Users/alejandrogarcia/Library/Arduino15/packages/arduino/tools/bossac/1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=/Users/alejandrogarcia/Library/Arduino15/packages/arduino/tools/bossac/1.6.1-arduino -verbose /Users/alejandrogarcia/Desktop/Restting Printer/Repetier-Firmware-4-Davinci/src/ArduinoDUE/Repetier/Repetier.ino
Using board 'arduino_due_x' from platform in folder: /Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8
Using core 'arduino' from platform in folder: /Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8
Detecting libraries used...
/Users/alejandrogarcia/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10812 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/libsam -I/Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/CMSIS/CMSIS/Include/ -I/Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/CMSIS/Device/ATMEL/ -I/Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/cores/arduino -I/Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/variants/arduino_due_x /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_401390/sketch/Repetier.ino.cpp -o /dev/null
Alternatives for SPI.h: [SPI@1.0]
ResolveLibrary(SPI.h)
  -> candidates: [SPI@1.0]
/Users/alejandrogarcia/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10812 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/libsam -I/Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/CMSIS/CMSIS/Include/ -I/Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/system/CMSIS/Device/ATMEL/ -I/Users/alejandrogarcia/Library/Arduino15/packages/arduino/hardware/sam/1.6.8/cores/arduino [... cut ...]
PORTS {/dev/cu.Bluetooth-Incoming-Port, /dev/tty.Bluetooth-Incoming-Port, } / {/dev/cu.Bluetooth-Incoming-Port, /dev/tty.Bluetooth-Incoming-Port, } => {}
Uploading using selected port: /dev/cu.Bluetooth-Incoming-Port
/Users/alejandrogarcia/Library/Arduino15/packages/arduino/tools/bossac/1.6.1-arduino/bossac -i -d --port=cu.Bluetooth-Incoming-Port -U true -e -w -v -b /var/folders/5c/8hlyxcd51_zbx5t9pjthb5yh0000gn/T/arduino_build_401390/Repetier.ino.bin -R 
No device found on cu.Bluetooth-Incoming-Port
Set binary mode
An error occurred while uploading the sketch
Send auto-baud
Set binary mode

luc-github, 2020-04-09:
the issue is in arduino ide, I do not know which version will work but here they write :
https://github.com/luc-github/Repetier-Firmware-4-Davinci/issues/206#issuecomment-258731165
1.6.12 is working with Sierra

The issue is clear `No device found on cu.Bluetooth-Incoming-Port` so port is invalid or you need some setup to open the port or may be some drivers to make printer recognize

closed 2020-05-02 by stale[bot]

timbroder, 2024-11-10:
@TexPonch did you ever figure this out?