luc-github/Repetier-Firmware-4-Davinci #361: [BUG] menu freezes and the printer restarts

KMC16, 2020-04-13:
Hello, I have a danvinci AIO, I have installed the program but the menu freezes and the printer restarts I just installed the program with the aurdiono 1.8.12 and the latest version of the github program, which may be the problem.

Thank you

luc-github, 2020-04-13:
wrong installation of the FW - you need to use the jumper to reset 
if you are not able to compile yourself you can use the binary avalable in release : https://github.com/luc-github/Repetier-Firmware-4-Davinci/releases

Light1964, 2020-04-13:
Hello and good Day 

 

Thanks for the quick reply

I forgot something when I saved for the first time

now he's running

Thanks again

 

Stay healthy

KMC16, 2020-04-14:
Yes I use jumper all the steps on the instructions, but I don’t know whit the program restar between 24 seconds

KMC16, 2020-04-14:
well, I have an AIO printer and I am doing the frimeware update, I am using the arduino ID 1.8.12 and due 1.6.8, I downloaded the files from github on 10 / April / 2020 and followed all the steps, the printer when I turn it on on the display it says REPETIER 0.91.10M, but the problem I have is that the printer restarts every 24 seconds

luc-github, 2020-04-14:
0.91.10?

KMC16, 2020-04-14:
[image]
Sorry 0.92.10M

luc-github, 2020-04-15:
this is watchdog not properly disabled, did you flashed several times with differents FW ?

KMC16, 2020-04-15:
Yes i do

KMC16, 2020-04-15:
I have another question I know is not on this topic but I would like to know how I can add the printer in the cura software

KMC16, 2020-04-15:
I am very grateful that you developed this FW this printer is very bad the truth I am that I bought another but well this is a very good solution, thanks for taking the time to develop this fw, I will try to do the process again to see if no longer restarts

luc-github, 2020-04-15:
the Watchdog flag can be tricky on this  board and several install may lock it
One solution is to disable the watchdog here : https://github.com/luc-github/Repetier-Firmware-4-Davinci/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L1580
`#define FEATURE_WATCHDOG 0`

for cura sorry I do not use for years, it but you just configure it like normal here a sample
https://www.xyzprinting.com/en-US/news/printing-with-cura
on 1.0A your bed size should 200 200 200 I think

closed 2020-05-11 by stale[bot]

Light1964, 2020-05-16:
thanks is all Ok