luc-github/Repetier-Firmware-4-Davinci #363: Firmware Installed Constant Restart

TheOfficialCzex, 2020-04-15:
The firmware seems to be installed (the LCD UI is functioning, the EEPROM was loaded...), but the printer keeps restarting every 30 seconds or so.

TheOfficialCzex, 2020-04-15:
My guess is that it's watchdog-based.

luc-github, 2020-04-15:
yes look like

TheOfficialCzex, 2020-04-15:
This is my first time dealing with anything Arduino-based, so I have no idea what I'm doing. How would I go about disabling watchdog or setting a loop trigger for it?

luc-github, 2020-04-15:
how did you opened issue ? there is a template for support

TheOfficialCzex, 2020-04-15:
Template?

luc-github, 2020-04-15:
yes when you open issue in github there is a template of questions to help for support.
I do not see them, so I am wondering how do you submit issue, you are not using github interface?

TheOfficialCzex, 2020-04-15:
I just created a new issue. Maybe you can't create an issue for your own release?

luc-github, 2020-04-15:
So when you create issue you do not see these ?

[image]

[image]

[image]

or do you open differently ?

TheOfficialCzex, 2020-04-15:
I think what happened was I started an issue without logging in, and, somehow, logging in between starting an issue and actually submitting it prevented me from seeing that page.

luc-github, 2020-04-15:
indeed - looks a bug in github
about issue now
 I need some info as you were not able to see the template:

**PC (please complete the following information):**
 - OS: [e.g. iOS]
 - Arduino Version [e.g.1.8.0]
 - Due Module Version [e.g. 1.6.8]

**Printer
- Type: [e.g. Duo 2.0, 1.0A, 1.0, AiO, Duo 2.0A]

**Firmware (please complete the following information):**
- Branch used [e.g. master branch]
- Version displayed in printer menu Settings/Sytem Version
- Configuration used in configuration.h
- Did you modified the code ?

TheOfficialCzex, 2020-04-15:
Windows 7
1.8.12
1.6.8
Master
0.92.10
0 (1.0A), 1 (J37), 0 (no added fan control)
Beyond the config, no.

luc-github, 2020-04-15:
Ok there is one possibility is that the 1.8.12 has some incompatibility - I used 1.8.0 and some more recent  and I had no issue, but I never used 1.8.12 for Davinci 

another solution is to disable the watchdog here : https://github.com/luc-github/Repetier-Firmware-4-Davinci/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L1580

`#define FEATURE_WATCHDOG 0`

and see if there is another problem with hardware

TheOfficialCzex, 2020-04-15:
Setting line 1580 to 0 completely fixed it. The printer runs without a hitch.

closed 2020-04-15 by TheOfficialCzex