luc-github/Repetier-Firmware-4-Davinci #365: When loading new filament mid print, filament unloads xy move to resume print.[BUG]

CraftedChannel, 2020-05-13:
**Describe the bug**
When you use DaVinci control panel to change filament mid print, everything works perfectly except as the head moves from the waste bucket/load filament position the feed stepper will run in unload direction.  It doesn't have to move very far before filament us fully unloaded.
**To Reproduce**
Steps to reproduce the behavior:
1. Start printing any object
2. After X layers, scroll on-printer controls to "Load Filament"
3. Hot pull filament
4. Insert new filament
5. Select on control panel "OK" to load filament.
6.  After load cycle completes, move cursor to "no" - press ok
7.  Print head moves in x and y back to resume print. During this move, the filament is unloaded by feed stepper.  Printing moves ensue but the filament is unloaded so no more filament is extruded.

**Expected behavior**
When print head moves back to resume print, Filament should not be extruded or retracted.  

**Screenshots**
If applicable, add screenshots to help explain your problem.

**PC (please complete the following information):**
 - Win 10
 - Correct for Repetier Davinci Firmware.
 - As provided with Davinci Firmware Upgrade.

**Printer
- Type: all 4 of them I own exhibit this bug.  I have 2x 1.0 and 2x 1.0A. Over 5000 hours print time cumulative.

**Firmware Repetier 0.92Mod
- Branch used  Stock installation downloaded a few years ago.  Not sure how to answer.
- Version displayed in printer menu Settings/Sytem Version 1.15-08-12_1.1A
Repetier o.92Mod
- Configuration used in configuration.h   I don't know how to find this.
- Did you modified the code ?  No

**Additional context**
I am a beginner here in GITHub.  I appreciate any influence I can have to improve this firmware.  I have much time using it and there are other bugs.  I figured I would start with this one and learn how to use GitHub.

luc-github, 2020-05-13:
you mean you are doing load filament without doing pause before ?

CraftedChannel, 2020-05-14:
Thank you, I will investigate along this line.  I did not realize there were newer versions available for the Davinici.

closed 2020-06-07 by stale[bot]

CraftedChannel, 2020-06-16:
Can you provide me a link to "1.17-08-01".  I don't know what it is I'm supposed to be looking for.  This number only appears here in our conversation and nowhere on github nor elsewhere on the internet according to google.

luc-github, 2020-06-16:
it is latest version on github