luc-github/Repetier-Firmware-4-Davinci #366: [FR] Can we have a updated V1 version ?

MasterCATZ, 2020-07-21:
https://www.repetier.com/firmware/v100/index.php

what settings are needed to recreate firmware for Da Vinci 2.0A Duo (j37)

I have been running 1.17-08-01_1.3A / 0.92.10M

luc-github, 2020-07-21:
there are very few difference for davinci between 1.x  and 0.92
also it is not matter of settings between otiginal repetier and the davinci version but code it self
original repetier FW cannot work on davinci printer

why do you need 1.x ?

MasterCATZ, 2020-07-23:
Thanks for reply 

I had not used mine for a few years and thought I would see about giving it an upgrade while I switch over to 
repetier-server (stopped using mine because I had to use SD card all the time because connection kept dropping out )

their were a few weird issues like it driving the head down hard on the plate etc  constantly trying to do head cleans and constantly getting left/right mixed up and another one was it unnecessary lifting up / down  when it was first turned on etc 

I was hoping some of this might have been corrected by now and was hoping auto level would be working

luc-github, 2020-07-23:
sorry what you describe looks not FW problems but something sending commands you do not expect or unexpected commands in slicer
latest version 0.92.10 is working fine and nothing of what you describe is fixed by repetier, the only open issue is : https://github.com/luc-github/Repetier-Firmware-4-Davinci/issues/265

I will works on porting repetier 2.0 when stable

luc-github, 2020-07-25:
The issue described is not related to FW so can we close issue ?

Bob-Rowles, 2020-07-25:
Yes, 

Thank you

 

Bob

closed 2020-07-25 by luc-github