luc-github/Repetier-Firmware-4-Davinci #370: [BUG]

JacobChrist, 2021-01-10:
**Describe the bug**
Non UTF-8 characters in source code.

**To Reproduce**
Opening the project several error messages like the following appeared:

"uimenu.h" contains unrecognized characters. If this code was created with an older version of Arduino, you may need to use Tools -> Fix Encoding & Reload to update the sketch to use UTF-8 encoding. If not, you may need to delete the bad characters to get rid of this warning.

Following the procedure to Tools -> Fix Encoding & Reload did not resolve the issue.

The code still complies, and this may not be an issue, but its scary because I don't want to brick my working printer.

**Expected behavior**
No errors upon opening the project.

**PC (please complete the following information):**
Windows 10
Arduino IDE 1.8.18
Arduino SAM Boards (32-bits ARM Cortex-M3): 1.6.8

**Printer
Duo 2.0

**Firmware (please complete the following information):**
- Branch used [master] 445fa50f8704c0ac70a9d769222763711cc16064
- Version displayed in printer menu Settings/Sytem Version
- Configuration used in configuration.h: Unmodified
- Did you modified the code: No

JacobChrist, 2021-01-10:
I tried to copy and past "uimenu.h" code into Notepad++ and convert to UTF-8 then delete and copy back to Arduino IDE, save the file and reopen the project and could not resolve this issue.

luc-github, 2021-01-10:
the readme mention the arduino version to use, did you read it ?

luc-github, 2021-01-10:
anyway you may workaround by disabling Japanese in configuration.h

`#define LANGUAGE_JP_ACTIVE 0 // Japanese`

JacobChrist, 2021-01-11:
> the readme mention the arduino version to use, did you read it ?

Fair enough.  I didn't use the same IDE, but it is a 1.8.x flavor and figured it would be okay.  The firmware compiles and loads onto my printer just fine.  Just wanted to log the issue.

luc-github, 2021-01-11:
I do not reproduce issue and I have tested the latest which is 1.8.13 , my windows support asian chars, so disabling japanese should remove the problem on your side,

closed 2021-01-11 by luc-github