luc-github/Repetier-Firmware-4-Davinci #371: Duo 2.0 X/Y location of print heads (both left and right) is over scraper while loading filament

JacobChrist, 2021-01-11:
**Describe the bug**
X/Y location of print heads (both left and right) is over scraper while loading filament.  This causes the newly loaded filament to extrude into the scraper gunking it up.

**To Reproduce**
Steps to reproduce the behavior:
1. From home screen press right arrow button.
2. Select Maintenance
3. Select Load/Unload
4. Select Load Left (or Right)

**Expected behavior**
Hot ends should be a bit further forward so as to clear the scrapers.

**PC (please complete the following information):**
Win 10
Arduino Version: 1.8.13
Due Module Version 1.6.8

**Printer
Duo 2.0 (no sure if its an 2.0A or not) w/J77
Main PCB Marking: 
F1.x F2.x Engine board F2
3DPO3-ES008
2014/08/01
Marked ICT, FCT1, VI

**Firmware (please complete the following information):**
Branch used [master] 445fa50
Version displayed in printer menu Settings/Sytem Version
Configuration used in configuration.h:
```
#define DAVINCI 3// "0" if not DAVINCI, "1" For DAVINCI 1.0, "2" For DAVINCI 2.0 with 1 FAN, "3" For DAVINCI 2.0 with 2 FAN, 4 for AiO (WITH NO SCANNER SUPPORT)
#define MODEL  1//"0" for first generation (jumper JP1 to reset ) , "1" for new generation   (jumper J37 to reset)
#define REPURPOSE_FAN_TO_COOL_EXTRUSIONS 0 //Setting this to 1 will repurpose the main Extruder cooling fan to be controlled VIA M106/M107
                                           //Warning: for DaVinci 1.0 need to add a permanent fan with power supply to cool extruder
```

Did you modified the code: No

luc-github, 2021-01-11:
I do not have Duo2A and position is based on homing position of my Duo2.0 but I know the scraper position changed with new generation 
Can you share a picture of the position of each extruders when home position - and when load/unload ?
also use the menu to move the extruder to put it in right position and give me the both positions ? So I can adjust the positions

JacobChrist, 2021-01-11:
The right extruder (labeled 1 on the printer) wants to be at X=0mm Y=30mm
The left extruder (labeled 2 on the printer) wants to be at X=195mm Y=30mm

Head over left extruder.
[image]
Close up with head over left extruder.
[image]
Head away from extruder after loading (black glob in scraper is PLA).
[image]
Main board picture:
[image]

P.S. Loving the firmware sidegrade!  The motion is so much smoother over stock firmware and it detects broken filament!

luc-github, 2021-01-11:
Ok Thank you - so I will add a define for the load / unload unload position in configuration.h according config
it should allow to correct position in case of problem - I will do this tomorrow

luc-github, 2021-01-12:
please give try to latest git - https://github.com/luc-github/Repetier-Firmware-4-Davinci/commit/1fcc1d041231c231bee23747710665f446f3ed55

JacobChrist, 2021-01-12:
Seems to work pretty good.  I may have been off, when the goop coming out of the hot end it looks like a y=25mm might be better.

luc-github, 2021-01-12:
you can adjust the values here https://github.com/luc-github/Repetier-Firmware-4-Davinci/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L1152-L1154 
SVP try and report me the best one

luc-github, 2021-02-02:
@JacobChrist did you finalized the proper x and y  ?

JacobChrist, 2021-02-05:
I will try to get some numbers over the weekend.

Jacob

JacobChrist, 2021-02-08:
Luc,

27 mm is the number I would go with.  25 mm is cutting is a bit close to
the scraper and 30 mm is close to the edge and when loading the printer the
extruded filament spills over the catch.

Jacob

luc-github, 2021-02-08:
done https://github.com/luc-github/Repetier-Firmware-4-Davinci/commit/84c107b10f49eb3ee85d6b531e7a6ad8c77850a3 thank you  ^_^

closed 2021-02-08 by luc-github