luc-github/Repetier-Firmware-4-Davinci #373: [Question]    How to Update Firmware after Conversion?

1RonRock, 2021-03-22:
I have a DaVinci 1.0A with Repetier 0.92.10 in working condition. I need to make some changes in the Configuration h. Temps and etruder motor direction. How do I save my current configuration to make these changes and not foul up my working config? Or how do I change my current firmware config with this current config? I already have Repetier on my printer.

luc-github, 2021-03-22:
I guess you are talking about your EEPROM settings
you can backup the EEPROM file on SD Card just in case
In theory if you only change temps / motor direction in Configuration.h the EEPROM won't be affected

1RonRock, 2021-03-22:
Wow thank you for the fast reply.

I have the Arduino 1.8.10 opened to the Configuration h tab. Have made my changes and Verified ok. My computer is connected to printer by USB cable. When I upload it runs the files but times out on upload. Receive Message(): timeout.

I have My Board set as Arduino Mega ADK, and Port set Arduino Due Com6 (correct com port)

Can I upload the current file found here? I only see a .bin file not sure how to use it.

1RonRock, 2021-03-22:
My configuration h shows define motherboard 33.

Is that correct motherboard? If I choose the DUE board and DUE port in Tools Board then I get an error that says that I must choose Arduino Mega as board.

I would much prefer to upload the current config and be done with my old configuration that I am sure has old settings that you have improved.

luc-github, 2021-03-22:
sorry did you already flashed this FW ?
Did you read the readme ?

1RonRock, 2021-03-22:
Yes it has been flashed.

Yes I have read the readme. Sorry it is very confusing to me.  From the readme it says that I do not need to follow steps 1 & 2. Step 3 says to use Arduino DUE 1.8.0. I am using 1.8.10 I assume is correct.

But I do not see how to load your current configuration. 

Sorry for my ignorance. I truly appreciate your help.

luc-github, 2021-03-22:
>I have My Board set as Arduino Mega ADK,

this is wrong , 

Also if you read the readme you will see some  files need from Due module to be modified

yes if already flashed 1 & 2 can be ignored
step 3 I do not know
All others steps are mandatories and strictly followed

1RonRock, 2021-03-22:
You seem to be getting upset with me. I am trying my best to follow the steps. I apologize again.

I thought that the board was incorrect. Thank you.

I have read and tried to follow the readme.

"4. Update variants.cpp/USBCore.cpp arduino files with the ones present in src\ArduinoDUE\AdditionalArduinoFile according your IDE version.
NOTE: You do not need to compile arduino from source these files are in the arduino directory structure FAQ#114.

5. Open the project file named repetier.ino located in src\ArduinoDUE\Repetier directory in the arduino IDE."

I have followed #114 and found the variants.cpp as well as the USB.cpp files. 

I do not see the src\ArduinoDUE\AdditionalArduinoFile so I do not have any file to use for update. 

I also do not see the repetier.ino also no  src\ArduinoDUE\Repetier directory in the arduino IDE.

luc-github, 2021-03-22:
I am not upset - no worry 

I have just highlighted you did not followed the readme - and emphazis to need follow strictly it - I really mean it
People usually just over read and think / or guess it is ok if they do not follow it - which always lead to problem 

Like select a Mega board when readme mention you must installl and use DUE module...

the files are presents in repository and you must see them
[image]

1RonRock, 2021-03-22:
Thank You. Yes I figured that the board choice was incorrect, that was the only choice that I was given to make the Arduino compiler work. I think that is because of the wrong motherboard choice in my current configuration h. Part of the reason I want to update to your more current and good configuration.

I did not understand that you were referencing the SRC in the repository. They are noted as "Update configuration h" that through me off. I will now find and change them.

1RonRock, 2021-03-22:
I think that I got it uploaded. It seemed to have worked. Although after leveling the bed I tried a print and the temp settings did not correspond to the settings I have set in my slicer. Same issue that I started with. I did find that I am able to change the temps in manual control and they do seem to stay in my choosen place, but the temp on the printer screen is still at an incorrect value. The printer bed temp shows 115, slicer is set at 100, manual control temp shows 110. Same as previously.

Is there a way to check if the firmware has actually been updated to the current one from the SRC in the repository?

Very frustrating! For both of us I'm sure.

luc-github, 2021-03-22:
change the version before flashing

I am not sure what temperatures are you refering too ? 
Printer do according gcode - so I am not sure to understand your issue between fw temps (which one) printer display and slicer  temp

1RonRock, 2021-03-22:
Sorry I failed to think that you were unaware of my temp setting issue. You have been very helpful, I much appreciate that.

I will do some more testing.

1RonRock, 2021-03-22:
> 
> 
> change the version before flashing in configuration.h
> 
> I am not sure what temperatures are you refering to?
> Printer do according gcode - so I am not sure to understand your issue between fw temps (which one?) printer display and slicer temp

Again thank you. I did change the last digit of the version and uploaded again. The number changed so I am confident that I now have the correct (new) firmware installed. And working. Now I have to figure out how to configure it for my E3D-V6 hotend and thermistor.

luc-github, 2021-03-23:
for that https://github.com/luc-github/Repetier-Firmware-4-Davinci/discussions/369

closed 2021-03-23 by luc-github