luc-github/Repetier-Firmware-4-Davinci #380: [FR]Add platformIO support

luc-github, 2021-04-13:
I will add platformIO support - it should allow to patch the due source automaticaly and so no need to to any manual changes 
also should allow to not change anything in configuration.h but use specific env for each config

So firmware should be easy to build in this way

luc-github, 2021-04-14:
PlatformIO support  is implemented but not tested - I have just checked the files are corrects
[image]

Patching happen once
[image]

I do not have bandwith to test right now  so if anyone want give a try and report would be great 😸 

Still need to modify src\ArduinoDUE\Repetier\Configuration.h according targeted printer 
I will add corresponding env in platformIO.ini once PIO build is confirmed as Ok

ybott, 2021-06-11:
I gave this a run on my DaVinci AiO. Build, upload, and monitor works.

Nice one!

luc-github, 2021-06-11:
great thank you for the feedback, I will add corresponding env

ybott, 2021-06-11:
I presume the work to integrate platformio is to merge all your work into the dev2 Repetier (https://github.com/repetier/Repetier-Firmware/tree/dev2) branch, is that correct?

luc-github, 2021-06-11:
everything is different in Davinci : specific screen , specific screen resolution( 16x4 ), specific SPI, specific features (auto bed leveling , manual bed leveling, dripbox, extruder cleaner, key pad, etc...

I will integrate repetier 2.0 when stable but push all davinci specific to repetier is big work and has few chance to be accepted, only some features may be

linpp, 2022-05-21:
On a Mac, this complains there are no source files to build.   It does build after changing src_dir to use forward slashes in platformio.ini:

src_dir     = src/ArduinoDUE/Repetier

hyx0329, 2022-10-28:
On my ArchLinux setup, this works smoothly with my davinci 1.0. Thank you for your work!

- platformIO core v6.1.4
- VSCodium 1.72.2
- XYZPrinting da Vinci 1.0(jumper J1)