luc-github/Repetier-Firmware-4-Davinci #390: Firmware resets when trying to "Upload to SD Card" from Simplify 3D

JacobChrist, 2021-11-23:
What is your question ?
Not sure what is going on here, but this just started.  When I try to upload a gcode to my printer from Simplify 3D the printer firmware reset and disconnects from the computer.  The SD card seems accessible: I can see gcode files from the printer menu and I can see that the new file is getting created.  Running the print from the computer seems to be working but I haven't tried a full print.  I don't think this is a bug per say but I wanted to document this here incase I find a solution or narrow the problem down.

[image]

The following is the output seen in the "Communication" tab of Simplify 3D:
```
SENT: M105
READ: ok 0
READ: T:42.22 / 0 B:40.13 / 0 B@:0 @:0 T0:42.22 / 0 @0:0 T1:30.28 / 0 @1:0
SENT: M28 spin2.gcode
SENT: G90
SENT: M82
SENT: M106 S0
SENT: M140 S60
SENT: M190 S60
SENT: M104 S190 T0
SENT: M104 S0 T1
SENT: M109 S190 T0
SENT: G28 
SENT: T0
SENT: G92 E0.0000
SENT: G1 E-1.0000 F1800
SENT: G1 Z0.200 F1000
SENT: G1 X44.954 Y60.788 F4800
SENT: G1 E0.0000 F1800
SENT: G92 E0.0000
SENT: G1 X46.301 Y59.792 E0.0501 F1350
SENT: G1 X47.727 Y58.812 E0.1019
SENT: G1 X49.093 Y57.938 E0.1505
SENT: G1 X50.543 Y57.079 E0.2009
SENT: G1 X52.014 Y56.273 E0.2511
SENT: G1 X53.512 Y55.518 E0.3014
SENT: G1 X55.004 Y54.828 E0.3506
SENT: G1 X56.598 Y54.155 E0.4024
SENT: G1 X58.163 Y53.558 E0.4525
SENT: G1 X59.748 Y53.014 E0.5027
SENT: G1 X61.351 Y52.523 E0.5528
SENT: G1 X62.969 Y52.088 E0.6030
SENT: G1 X64.548 Y51.719 E0.6515
SENT: G1 X66.245 Y51.382 E0.7033
SENT: G1 X67.894 Y51.114 E0.7533
SENT: G1 X69.558 Y50.901 E0.8036
SENT: G1 X71.210 Y50.746 E0.8532
SENT: G1 X72.938 Y50.644 E0.9050
SENT: G1 X74.558 Y50.602 E0.9536
SENT: G1 X76.234 Y50.616 E1.0037
SENT: G1 X77.909 Y50.686 E1.0539
SENT: G1 X79.633 Y50.818 E1.1057
SENT: G1 X81.298 Y51.003 E1.1558
SENT: G1 X82.956 Y51.245 E1.2060
SENT: G1 X84.586 Y51.539 E1.2555
SENT: G1 X86.226 Y51.892 E1.3058
SENT: G1 X87.851 Y52.301 E1.3559
SENT: G1 X89.513 Y52.781 E1.4077
SENT: G1 X91.055 Y53.282 E1.4562
SENT: G1 X92.630 Y53.853 E1.5064
SENT: G1 X94.235 Y54.499 E1.5582
SENT: G1 X95.718 Y55.155 E1.6067
SENT: G1 X97.227 Y55.883 E1.6569
SENT: G1 X98.758 Y56.688 E1.7086
SENT: G1 X100.183 Y57.501 E1.7577
SENT: G1 X101.611 Y58.380 E1.8080
SENT: G1 X103.009 Y59.308 E1.8582
SENT: G1 X104.340 Y60.258 E1.9071
SENT: G1 X105.714 Y61.309 E1.9589
SENT: G1 X107.008 Y62.372 E2.0090
SENT: G1 X108.266 Y63.480 E2.0592
SENT: G1 X109.446 Y64.592 E2.1077
SENT: G1 X110.665 Y65.819 E2.1595
SENT: G1 X111.840 Y67.089 E2.2113
SENT: G1 X112.902 Y68.318 E2.2599
SENT: G1 X113.956 Y69.623 E2.3101
SENT: G1 X114.982 Y70.987 E2.3612
SENT: G1 X115.943 Y72.360 E2.4114
SENT: G1 X116.828 Y73.718 E2.4599
SENT: G1 X117.723 Y75.199 E2.5117
SENT: G1 X118.539 Y76.662 E2.5619
SENT: G1 X119.282 Y78.104 E2.6104
SENT: G1 X120.021 Y79.667 E2.6622
SENT: G1 X120.685 Y81.206 E2.7124
SENT: G1 X121.277 Y82.715 E2.7609
SENT: G1 X121.851 Y84.337 E2.8124
SENT: G1 X122.368 Y85.979 E2.8639
SENT: G1 X122.803 Y87.541 E2.9125
SENT: G1 X123.198 Y89.169 E2.9626
SENT: G1 X123.547 Y90.864 E3.0144
SENT: G1 X123.821 Y92.462 E3.0629
SENT: G1 X124.049 Y94.122 E3.1131
SENT: G1 X124.224 Y95.843 E3.1649
SENT: G1 X124.334 Y97.461 E3.2134
SENT: G1 X124.392 Y99.135 E3.2636
SENT: G1 X124.393 Y100.839 E3.3146
SENT: G1 X124.336 Y102.516 E3.3648
READ: wait
READ: ok 0
READ: Writing to file: spin2.gcode
READ: ok 0
WARNING: Device unplugged while connected to port
Total build time: 0.16 minutes
Disconnected.
```

luc-github, 2021-11-24:
some boards reset when starting serial connection , may be once finished sending file s3D do a new connection so it reset board

JacobChrist, 2021-11-26:
It doesn't always happen, sometimes I can upload a file fine.  I haven't done enough testing but it could be my process in s3D generating bad g-code.  When it happens the whole thing resets: all the lights shut off and the motors holding current is released.

luc-github, 2021-11-26:
the usb cable can be subject to emi noise if close to any electric source, another possible reason to restart is the watch dog due to serial transfert is somehow stuck

closed 2022-01-09 by stale[bot]