luc-github/Repetier-Firmware-4-Davinci #391: [BUG]

torp105, 2021-12-18:
Sorry If I am missing anything. I am new to github.

I am having some troubles with using this firmware with my printer. I am using the davinci 2.0A Duo. I think, I got the firmware to upload correctly. Quite a few things don't work. The stepper moters for the x, y, and z seem to work, but the limit switches only work for x and y, the lights don't work, and the heated elements don't work or report at all.

Sorry If I don't have enough info here for your to work with. Let me know if there is any other info that I can give you to help.

Thanks

luc-github, 2021-12-19:
what is version display in system menu?
did you loaded failsafe eeprom settings after flash?

luc-github, 2021-12-29:
@torp105  Any update?

torp105, 2022-01-02:
Sorry about the late reply. Just coming back from christmas break.

To answer your question the version is: 
Da Vinci 1.0
1.21-02-08_1.1
Repetier 0.92.10M

torp105, 2022-01-02:
oh yes and I did load failsafe eeprom settings after flash

luc-github, 2022-01-02:
you flashed davinci 1.0 but you mention you have a davinci 2.0A so there is a problem

torp105, 2022-01-03:
OH! so... did I do something wrong in the Arduino coding section?

luc-github, 2022-01-03:
I guess, because result is Davinci 1.0 instead of 2.0A and pins are differents so it may explain why is not moving

torp105, 2022-01-03:
What should I do?

luc-github, 2022-01-03:
edit configuration.h and modify according 2.0A 

check readme , it explain what to change in configuration.h

torp105, 2022-01-04:
Ok. I will try to do that today or tomorrow. But, I could have sworn that I did do that, but if the machine is saying I did it wrong then I must have done it wrong

torp105, 2022-01-05:
Looks like that fixed it! I guess I just forgot to change something in configuration H. Now all I need to do is figure out how to get cura or some other slicer to work with it

Sorry for wasting your time and thank you for your help!

closed 2022-01-05 by torp105