luc-github/Repetier-Firmware-4-Davinci #392: Hi Correct me if I'm wrong, I found this typo FAN_PIN2 should be FAN_PIN.

johnnysmithy852456, 2022-01-30:
https://github.com/luc-github/Repetier-Firmware-4-Davinci/blob/c6cd7a4b72a4a05405694e6eebb3e015aae464ec/src/ArduinoDUE/Repetier/Configuration.h#L1921

johnnysmithy852456, 2022-01-30:
Hi Correct me if I'm wrong, I found this typo FAN_PIN2 should be FAN_PIN. I was trying to get the second part cooling fan running would not work until I change it to FAN_PIN.

luc-github, 2022-01-30:
No this code is from original repetier FW and it is not used as you see it is commented 
the code used is here : 
```
#define REPURPOSE_FAN_TO_COOL_EXTRUSIONS 0 //Setting this to 1 will repurpose the main Extruder cooling fan to be controlled VIA M106/M107
                                           //Warning: for DaVinci 1.0 need to add a permanent fan with power supply to cool extruder
```
 and here
https://github.com/luc-github/Repetier-Firmware-4-Davinci/blob/c6cd7a4b72a4a05405694e6eebb3e015aae464ec/src/ArduinoDUE/Repetier/Configuration.h#L1902

johnnysmithy852456, 2022-01-31:
Hi Luc
Isn't the repurpose option for people that want to change the function of the Extruder cooling Fan to a g-code switchable part cooling fan.
I don't want to change the function of the original fan, I only wanted to enable the second fan on Pin 85 on the Divinci 1.0A.

I finally got it working after adding Pin 85 to the userpins.h file.

Uncommenting line 1921 #define FAN2_PIN ORIG_FAN2_PIN

And changing FAN2_PIN to FAN_PIN

Now I have a new setting in the top menu called Fan Speed.

3.3v when the g-code is sent and about half that if I call for 50% speed.
Thanks for getting back to me on this.

luc-github, 2022-01-31:
Yes it is - but as free pins are not easily accessible - this case is not currently set - where is pin 85 located on 1.0A ?May I know  how did you wired the fan ?

luc-github, 2022-02-11:
wow nice, thank you for sharing

luc-github, 2022-02-11:
I have renamed the issue for better search for anyone interrested in same feature

closed 2022-02-11 by johnnysmithy852456

leeroyjenkins-87, 2023-12-08:
> Hi Sorry about the late reply, I've managed to test this with a PWM fan and Ive markup a drawing for you to follow.
> 
> The 3.3v PWM signal (GPIO85) from the MCU will not drive the fan at full speed because the fan expects to see a 5V PWM signal. Thats why I've added a Logic level converter (BOB-12009) Any logic level converter should work.

Hi Johnny, I have managed to repeat the hardware process you had mentioned. I got close to having repetier send the Gcode. But I am not getting the fan to turn on. If I had to guess its my configuration for pin 85... Been trying a couple things. this is what I currently have. 

In Cofiguration.h:
#define REPURPOSE_FAN_TO_COOL_EXTRUSIONS 1

And 

/* You can have a second fan controlled by adding P1 to M106/M107 command. */ 
#define FEATURE_FAN2_CONTROL 0
#define FAN2_PIN ORIG_FAN2_PIN //WAS //#define FAN2_PIN ORIG_FAN2_PIN
#define FAN_PIN ORIG_FAN2_PIN //WAS //#define FAN2_PIN ORIG_FAN2_PIN

in Userpins.h:

//fan PINS
#define FAN_BOARD_PIN -1
#define ORIG_FAN_PIN        25 //WAS 25
#define ORIG_FAN2_PIN       85 //WAS 4

I was looking at the pinouts for the chip on the board and the area you say to soldier to was a pin # around "175-185" went searching the chip schematic looking for GPIO85 somewhere in that region and didn't find anything. So I'm wondering if I got the wrong pin number (85) in the config.