luc-github/Repetier-Firmware-4-Davinci #393: Getting an error when I verify, am I doing something wrong?

fredsterrr, 2022-02-02:
I'm trying to flash my Davinci with Repetier .92, I've followed the steps to the best of my ability, but when I click verify on Arduino (1.5.8) the following message appears:

Arduino: 1.5.8 (Windows 8), Board: "Arduino Due (Native USB Port)"

Using library SPI in folder: C:\Users\fred.trevvett\Downloads\arduino-1.5.8-windows\arduino-1.5.8\hardware\arduino\sam\libraries\SPI 

C:\Users\fred.trevvett\Downloads\arduino-1.5.8-windows\arduino-1.5.8/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -mcpu=cortex-m3 -DF_CPU=84000000L -DARDUINO=158 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON -DUSB_MANUFACTURER="Unknown" -DUSB_PRODUCT="Arduino Due" -IC:\Users\fred.trevvett\Downloads\arduino-1.5.8-windows\arduino-1.5.8\hardware\arduino\sam\system/libsam -IC:\Users\fred.trevvett\Downloads\arduino-1.5.8-windows\arduino-1.5.8\hardware\arduino\sam\system/CMSIS/CMSIS/Include/ -IC:\Users\fred.trevvett\Downloads\arduino-1.5.8-windows\arduino-1.5.8\hardware\arduino\sam\system/CMSIS/Device/ATMEL/ -IC:\Users\fred.trevvett\Downloads\arduino-1.5.8-windows\arduino-1.5.8\hardware\arduino\sam\cores\arduino -IC:\Users\fred.trevvett\Downloads\arduino-1.5.8-windows\arduino-1.5.8\hardware\arduino\sam\variants\arduino_due_x -IC:\Users\fred.trevvett\Downloads\arduino-1.5.8-windows\arduino-1.5.8\hardware\arduino\sam\libraries\SPI C:\Users\FRED~1.TRE\AppData\Local\Temp\build7703501508648193387.tmp\BedLeveling.cpp -o C:\Users\FRED~1.TRE\AppData\Local\Temp\build7703501508648193387.tmp\BedLeveling.cpp.o 

In file included from C:\Users\FRED~1.TRE\AppData\Local\Temp\build7703501508648193387.tmp\Repetier.h:572:0,
                 from C:\Users\FRED~1.TRE\AppData\Local\Temp\build7703501508648193387.tmp\BedLeveling.cpp:107:
C:\Users\FRED~1.TRE\AppData\Local\Temp\build7703501508648193387.tmp\HAL.h: In static member function 'static void HAL::serialSetBaudrate(long int)':
C:\Users\FRED~1.TRE\AppData\Local\Temp\build7703501508648193387.tmp\HAL.h:684:14: error: 'class UARTClass' has no member named 'setInterruptPriority'
       Serial.setInterruptPriority(1);
              ^
Error compiling.

Is this something I'm doing wrong, or an error with the files on Github?  I haven't touched the files it mensions, so it looks like something is wrong with them from the start, as opposed to me doing something wrong?  Happy to be proven wrong on tha, I just want an answer!!

Thanks in advance

Fred

luc-github, 2022-02-02:
did you read the read me telling that using 1.5.8 will give you compilation error?https://github.com/luc-github/Repetier-Firmware-4-Davinci#installation

fredsterrr, 2022-02-02:
Hello @luc-github,

Yes I did, but...
First I tried IDE 1.8.19 and it couldn't connect to the DUE
So I tried 1.5.8 because there was a YouTube tutorial and the guy was using that version and it worked fine, however that was a few years old

To be honest, I don't care what version or method I use, I just want to get repetier on my 3d printer and get it back in action!  I'll try with 1.6.8 and report back

luc-github, 2022-02-02:
the readme is the one which work because is the one which follow  code update

the video is obviously outdated

fredsterrr, 2022-02-02:
Sorry I should say I'm completely new to Arduino, completely new to any form of tinkering with software really, the most similar thing I've ever done is installed chrome on a new computer😁

So I'm sorry if some of what I'm doing seems a bit thick, im afraid you will have to bear with me!!

fredsterrr, 2022-02-02:
Ok thanks, 1.6.8 installing now...

fredsterrr, 2022-02-02:
Hmm, where is the variants.cpp file in the Arduino files?  I can find the one in the repetier files, but can't find where to move it to (step 4)

fredsterrr, 2022-02-02:
Ah I've just realised I've downloaded repetier 1.0.0, is that a problem?  As far as I can see it shouldn't be.  The problem I'm having (not finding files in step 4) relates to the Arduino files, not the repetier files so so I don't think it's an issue.  However if I do need to download 0.92 I can do, I just don't like using software when a new version has been released since

luc-github, 2022-02-02:
this repository is a repetier version for Davinci printer, it is heavily modified from original to work with Davinci hardware
if you go to main repetier repository (that does not support davinci hardware) and try to use it on your davinci it won't work, that is why my repository exists

fredsterrr, 2022-02-02:
Sorry - ignore my last comment, not sure what I was thinking, I am working with your version.  I downloaded 1.0.0 a few days ago before I realised I needed your version.

Do you know where I will find variants.cpp in the Arduino files?

luc-github, 2022-02-02:
in step 4;there is a link about it , click on it and read the FAQ item

fredsterrr, 2022-02-02:
I looked there but the files don't match what I've got on my computer, I got as far as the appdata folder but there was no Arduino folder in there

luc-github, 2022-02-02:
did you tried the method mentioned ?
https://github.com/luc-github/Repetier-Firmware-4-Davinci/discussions/114#discussioncomment-202662

fredsterrr, 2022-02-02:
Sorry no, I read that as 'relevant to all iOS' so I skipped it as I am on windows.  That took me to the right place, I'm just uploading it now...

fredsterrr, 2022-02-02:
€25 on its way to you, all the best👍👍👍

luc-github, 2022-02-03:
@fredsterrr thank you for your kind words and your support -  happy printing