luc-github/Repetier-Firmware-4-Davinci #396: Davinci 1.0 AiO Black Bars after SD card removal

TomD66069, 2022-04-10:
I have a davinci 1.0 AiO that I've had repetier on for the last few months with no issues (uploading prints by USB). I had to take my computer with me to work last week and was going to start a print through the SD card.

I removed the SD card, uploaded the gcode and put the card back in the printer. When I turned the printer back on it had the black bars, just like after flashing the fw. 

I have reformatted the sd card and copied the backups i made of the files back onto the card and that didnt work. I cleaned the sd card and card slot, still not working. Tried a new sd card, didnt work. 

I tried to plug my computer back in and go through re-flashing the printer but when i plug the printer in I get no feedback. In arduino the ports tab stays grayed out, and in device manager it says there are no usb devices connected. I even got the windows usb device viewer to try and debug the port but it still said nothing was connected. 

Is there anything else that I can try or is my printer bricked?

luc-github, 2022-04-10:
It looks like you got a short when removing the SD that damaged the flash could be static electricity also when you touched the board

first thing is to make your printer detected
1 - use jumper to reset the board  (J37 for AIO) and do it slowly 
if your board is still not detected by window when you plug USB - I am afraid that there is not so much to do but start to change the board by a new one

TomD66069, 2022-04-10:
I already tried following your instructions from a different thread to jump it for 10 seconds then do it another time to be sure. No luck with that either. 

My guess if I'm going to have to change the board. Is there somewhere that I can look into getting a new board? I have no idea where to shop for something like that (preferably a Canadian site if you know one)

luc-github, 2022-04-10:
No sorry - ebay is the best chance I guess - Davinci boards are pretty fragile due to the cost down company is doing

My suggestion would be to think about a mod changing the main board if you cannot find a replacement board

TomD66069, 2022-04-10:
> No sorry - ebay is the best chance I guess - Davinci boards are pretty fragile due to the cost down company is doing
> 
> My suggestion would be to think about a mod changing the main board for real printer board if you cannot find a replacement board

Okay thanks for the info!

Is there a thread on here for doing that or is that something I'd need to find elsewhere?

luc-github, 2022-04-11:
I would go here : https://www.soliforum.com/forum/21/xyz-printing-davinci/
and https://www.thingiverse.com/groups/da-vinci and google of course ^_^