luc-github/Repetier-Firmware-4-Davinci #401: Hotend specs?

fredsterrr, 2022-07-25:
(Posted first in Discussions but no reply)

I replaced my hotend as the old one was faulty, I got a 12v 40w one but now it heats, but not hot enough. It gets to about 150deg and then stops, after a minute or so it then triggers the thermal runaway protection.

I'm guessing I got the wrong voltage or wattage, can someone please tell me what spec hotend the firmware is designed to work with so I can buy the correct one?

Thanks.

luc-github, 2022-07-25:
I remember it was 12V but I do not have system anymore , if you changed the hothend do it means you changed the thermistor because your issue is more likely that

fredsterrr, 2022-07-25:
Yes I did change the thermistor, but I think that's working properly.  It reads room temperature correctly, and when I heat it up to 150 then use manual g-code to force filament through, it doesn't extrude any filament.  I also have a cheap infrared thermometer which shows a very similar temperature on the hotend, to what the thermistor shows.

I still have all the files on my computer from when I flashed the printer, is there any way I can open these up and read the settings manually?  Where would I need to look?

fredsterrr, 2022-07-25:
Ok thanks Luc.