luc-github/Repetier-Firmware-4-Davinci #402: [BUG] Freezes and resets every 10-30 seconds ad infinitum

KeystrokeCascade, 2022-10-01:
**Describe the bug**
Immediately after flashing and powering on the printer, it works perfectly as expected for 10-30 seconds, before freezing for about 5 seconds, then resetting.  It retains no settings when doing so.  Happens when both connected to a computer or not.

**To Reproduce**
Not sure it's reproducible but 
Steps to reproduce the behavior:
1. Power on printer
2. (Optional) Just scroll up and down the menu so you can see when it stops responding to input
3. Wait 10-30 seconds for it to freeze then reset

**Expected behavior**
It doesn't do that.

**PC (please complete the following information):**
 - OS: Arch Linux
 - Arduino Version: 1.8.19-2
 - Due Module Version: 1.6.8

**Printer**
- Da Vinci 1.0

**Firmware (please complete the following information):**
- Branch used: Master
- Version displayed in printer menu Settings/System Version: 1.21-02-08_1.1 / Repetier0.92.10M
- Configuration used in configuration.h: `DAVINCI 1`, `MODEL 0`, (also tried with model 1 just in case)
- Did you modified the code?: No

**Additional context**
Also tried switching power supplies and with the back panel on and off in case it was overheating.  Took a log of the printer by running `cat /dev/ttyACM0 > printer.log`, didn't see anything weird.

Before flashing it I made sure it was working with the factory firmware after I levelled the bed, and it did successfully complete a test print and I never noticed any such issues.

luc-github, 2022-10-01:
this looks like watchdog issue - disable the watchdog to see what is wrong https://github.com/luc-github/Repetier-Firmware-4-Davinci/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L1621

KeystrokeCascade, 2022-10-01:
Thanks! It seems like disabling the watchdog has fixed it.  I'm not sure about the SD card, I haven't touched it since I got the printer and while it seems too cramped to pull out and push back in without removing the mainboard, I did make sure it was pressed in.  Any adverse effects from disabling the watchdog other than if it should freeze up again, I might have to manually power-cycle it?

An interesting observation is that with the watchdog disabled, it still froze after about 20 seconds, however after 5 more seconds resumed with no problem and hasn't frozen since.  Is there a watchdog timeout that could be altered or something that runs in the background after 10-30 seconds that may cause this hiccup?

luc-github, 2022-10-01:
no, the freeze generate the watch dog reset, without watdog the freeze still happen but does not reset the printer

Disabling the watch dog is not the solution - it is just a work around to see what freeze your printer, without watchdog if another freeze hapen when you are heating your printer may loose control of heating and so bad things will happen 

check if SD card is writable by doing the save EEPROM settings - I only see that problem that may cause some freeze as EEPROM emulation is actualy using SDCard

KeystrokeCascade, 2022-10-01:
Doing the EEPROM save in the settings at the end of the setup tutorial worked, however during an attempt to add a USB power supply to the mainboard I accidentally shorted 12v to 3.3v and killed the board, so now I'm also taking suggestions for replacements for that.  Sigh...

luc-github, 2022-10-01:
Sorry to read this, well Davici board is very fragile but from positive point of view you will get better result by changing board 

Changing board itself won't be the most difficult - changing endstops and screen is the most difficult - there are plenty of tutorials like https://www.techmonkeybusiness.com/articles/Frankenvinci.html

I personnaly suggest you to not use ramps but stm32 or even esp32 to get wifi support - they are cheap,
 and I also suggest to use a serial tft screen for the text screen replacement - just my 2 cents

KeystrokeCascade, 2022-10-02:
Trust me, nobody's more sorry about this than me.  Thank you so much for the help and the recommendation, I'll be sure to check them out.

As for what was wrong with the mainboard I guess it took that secret to the grave so there's no point in keeping this open.

closed 2022-10-02 by KeystrokeCascade