luc-github/Repetier-Firmware-4-Davinci #403: [Question]<XYZ Da Vinci 2.0 Duo X Axis Seems to be off, hits left wall and continues click and stutter.

CaptainFerret, 2022-11-13:
I'm pretty new to 3D printing and received a good price on a XYZ Da Vinci Duo that ended up being modded with Repetier 0.92. 10M. I've been feeling my way through calibrating and leveling the bed, and it seems like the X axis is shifted farther to the left than it should be. Whenever I start a print, the extruder will move as far left as it can, then starts stuttering because it wants to go further. From there, it will print on the floor of the printer rather than the bed. I have some other questions that may be basic, but I mostly want to fix this issue.

- Where is X0,Y0 supposed to be located on this printer
- How can I properly calibrate the X0,Y0,Z0 in both the printer itself and Repetier Host
- In the future when I want to use the second Extruder, how much should I offset the second extruder? 
- Do I have to program both dump bins for the Load/Unload function

Thank you very much for your help, and I'm sorry if some of these questions are very basic. Have a great day!
[image]
 
This is the side it will move to while trying to clean the extruders

luc-github, 2022-11-14:
* x0y0 should be on print bed close to home position
* because home position are xmin ymin and zmin the 0 are related to these positions
* the offset are already set for second extruder based on my duo, that said.you may still need to adjust them , I personnaly used the traffic cone object from thingverse

* clean extruders command https://github.com/luc-github/Repetier-Firmware-4-Davinci/blob/master/src/ArduinoDUE/Repetier/Repetier.ino#L118
* load unload filament are set directly in printer menu not sure to understand your question
*

CaptainFerret, 2022-11-14:
So, once I define the xyz min positions, the printer will find it's own home based on that?
Is there a particular method that's better for finding the xyz min positions? Right now I'm been trying to feel my way through the process to no success
Sorry if these are basic questions, and thank you very much for your help!

luc-github, 2022-11-14:
no the home position is always same it use the end stops

luc-github, 2022-11-14:
by default if settings for xmin ymin are correct check using only printer menu and printer jog
to edit values,you can use repetier host, there is documentation on repetier for this

CaptainFerret, 2022-11-16:
I'm not sure if this may be a problem, but when I home Y and Z, it accurately travels to the end stops. However, when I home X, it will move about 50mm left, and each time I press home it will continue moving in 50mm steps until reaching the left rail and jamming. I've realized I can disable the stepper and move it to the X end stops by hand, and it'll print somewhat normally, however at some point it will begin stuttering into the left rail. 

Could this be a hardware issue or the previous owner playing with settings and messing it up?
The XYZ min settings are also all set to 0 for some reason on my printer, and I believe that it came this way from the previous owner. 
Thank you very much for your help, you're truly making this a better experience for me!

CaptainFerret, 2022-11-16:
I can supply pictures or videos as examples if needed

luc-github, 2022-11-16:
I am sorry but if you have no idea of the settings used to flash the FW I cannot really help you to know if it is hardware or setting issue.

I would suggest to flash FW by yourself to do a clean start and do a load failsafe Firmware to clear old settings