luc-github/Repetier-Firmware-4-Davinci #410: What am i doing wrong?

Timboslice8, 2023-10-25:
I need help! iv been at this for days, what am i doing wrong? 

To give you some back ground on my journey, I got a used Da Vinci 1.0A unit and decided to try and make it work.  I started off with the Repetier .91 build but i soon came to realize that would not work for me as i have the new Da Vinici model with J37
Im now attempting to install the .92 firmware and iv tried troubleshooting this every which way i can think of.   I was making a connection with Da Vinci unit initially and then i realized i forgot to copy the hardware folder as my printer was not being recognized after installing was complete as per point 4.  Then once i copied the files to my Arduino folder i started getting this error.  I tried uninstalling and reinstalling Arduino and even tried different versions to no avail. Any thoughts?

Arduino: 1.8.0 (Windows 10), Board: "Arduino Due (Native USB Port)"

Board arduino_due_x (platform sam, package arduino) is unknown

Error compiling for board Arduino Due (Native USB Port).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

luc-github, 2023-10-25:
you did copy wrong for sure based on your description, copy in wrong place may be , using wrong Due Core Module may be
`every line` and I repeat `every line` in  read me is key and mandatory

You must  redo everything from scratch, delete everything including the arduino due module and reinstall everything carefully 

You could also try to use platformio I have automated everything in it so you should not worry about module version neither patch

Timboslice8, 2023-10-26:
Ok so i will redo everything as per your recommendations.
Step 1 )I have Jumped the Eprom from J37.  Device manager is showing Boss Program (COM7)
Step 2) I have uninstalled and reinstalled Arduino 1.8.0. Downloaded from https://www.arduino.cc/en/software/OldSoftwareReleases
Step 3) I have Re-Downloaded and Unpacked Repertier Firmware from https://github.com/luc-github/Repetier-Firmware-4-Davinci
Step 4)a) Opened C:\Users\Timo\Desktop\Repetier-Firmware-4-Davinci-master\Repetier-Firmware-4-Davinci-master\src\ArduinoDUE\AdditionalArduinoFiles\Arduino - 1.8.0 -Due 1.6.8 where the "hardware" folder is
Step 4)b) Opened C:\Users\Timo\AppData\Local\Arduino15\packages\arduino  
Step 4)c) Dragged "hardware" folder from step a to folder in step b -
 note: it did not copy over anything.  In my sam folder there is a 1.6.11 and a 1.6.12

Please Advise

Timboslice8, 2023-10-26:
[image]

luc-github, 2023-10-26:
I do not see the step you installed Due module, did you ? the 1.6.8 as mentionned in read me and described in files path to modify

Timboslice8, 2023-10-26:
i did not , but i have now.  I copied the files and they have replaced the ones as expected.
Step 5) Opened the project file named repetier.ino located in src\ArduinoDUE\Repetier directory
Step 6) Modified the DAVINCI define in Configuration.h file.  Left top as "1" changed middle to "1" and left bottom as "0"
Step 7) Ensured I had Arduino DUE (Native USB Port) selected as well as COM7
Step 8) Pressed compile and received error message

Arduino: 1.8.0 (Windows 10), Board: "Arduino Due (Native USB Port)"

Board arduino_due_x (platform sam, package arduino) is unknown

Error compiling for board Arduino Due (Native USB Port).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Timboslice8, 2023-10-26:
I just figured it out.. The firmware was installed successfully!
So last night out of desperation i started copying the variants.cpp/USBCore.cpp arduino files to other places.... 
When you uninstall Arduino it does not remove all the files under C:\Users\Timo\AppData\Local\Arduino15\packages\arduino so i had to manually delete these files to completely start fresh. 
Thank you for your help! Your ongoing dedication to this project is inspiring.

Best Regards,

Tim

luc-github, 2023-10-26:
happy to see you managed it :+1: , happy printing

closed 2023-10-26 by luc-github