luc-github/Repetier-Firmware-4-Davinci #411: How to add 2nd fan to DaVinci 1.0a

skinnyguyfpv, 2023-10-28:
Hi guys, I know this is an old printer, but after flashing repetier on my 1.0a I've printed tpu, pla, petg and prints very well, I want to add a 2nd fan and I know there is and older post about doing it but I need some help in doing it, where abouts do I put #define add extra fan etc? Do I put it in Arduino? And reflash the computer? I'm confused, please help, everything is soldered up ready go

luc-github, 2023-10-29:
the components on board 1.0/a are not present for a second fan

The easiest solution is to power your extruder fan  externly and to repurpose the pi of current fan for fan control

skinnyguyfpv, 2023-10-30:
Thanks for the reply, I have the same printer PCB as your photo with fan B though, do I need to reset the printer or just verify and write in Arduino? I'm just worried about it resetting all my print settings I have

luc-github, 2023-10-30:
there is not need to reset any more - just change from your current  Configuration.h for 1.0a

https://github.com/luc-github/Repetier-Firmware-4-Davinci/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L26
And reflash