luc-github/Repetier-Firmware-4-Davinci #416: After uploading "usb device not recognized" constantly reconnecting printer still displaying 2 black bars

Kitzakip, 2024-05-17:
What is your question ?

Kitzakip, 2024-05-17:
using arduino 1.8.0 with the 1.6.9 sam board manager version, ive swapped out the hardware folder too

luc-github, 2024-05-17:
Why using SAM 1.6.9? the https://github.com/luc-github/Repetier-Firmware-4-Davinci?tab=readme-ov-file#installation
is very clear - and it also mention what to do if you have such situation 
Every line of this readme is important - every line - read carefully

Kitzakip, 2024-05-17:
ah my mistake that was a typo i am using SAM 1.6.8, its step 4 i think im having issues with, ive updated the variants.cpp/USBCore.cpp arduino files under C:\Users\Quilla\AppData\Local\Arduino15\packages\aurduino\hardware is that correct? ive seen a few different posts with people updating them in appdata vs program files and the install directory itself

luc-github, 2024-05-17:
As wrote everything is in Readme : https://github.com/luc-github/Repetier-Firmware-4-Davinci/discussions/114
Please read carefuly

kalitakomwa, 2024-06-10:
im having the same issue, usb device not reckognized - spamming and printer showing two bars. i cant do step 4 again because it doesnt recognize printer as port, tried flashing again jp1 to no avail

Kitzakip, 2024-06-10:
ive since fixed it though luc was right, fresh install in a new directory and verryyyyy carefully reading the readme fixed it for me