luc-github/Repetier-Firmware-4-Davinci #417: I think my printer isn't being detected? da vinci 1.0

kalitakomwa, 2024-06-10:
went through all the steps, trying to now press on upload on the Arduino, got this error.. anyone know why? 
[image]

luc-github, 2024-06-10:
what ports are visibles when you go to tools->ports ?

luc-github, 2024-06-10:
your printer is not detected redo the operation with the jumper and check it is then detected by window

kalitakomwa, 2024-06-10:
Oh wait i see something now!

kalitakomwa, 2024-06-10:
[image]

kalitakomwa, 2024-06-10:
i think it's working now! been stuck on this right now for some time
[image]

kalitakomwa, 2024-06-10:
it seems to be stuck on this line :(