luc-github/Repetier-Firmware-4-Davinci #418: Script not Compiling

emrindustries, 2024-12-13:
Hello Everyone 

I cant get the script to comply it comes up with these errors... My printers usless now as ive flashed the firmware to nothing any help would be great.
I did the same mod years ago without any problems but there has been updates i think and its different to the first time i did it 7 or 8 years ago.. That printer is still going strong this is a new one i picked up..  

In file included from Repetier.h:572:0,
                 from BedLeveling.cpp:107:
HAL.h: In static member function 'static void HAL::serialSetBaudrate(long int)':
HAL.h:684:14: error: 'class UARTClass' has no member named 'setInterruptPriority'
       Serial.setInterruptPriority(1);
              ^
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

luc-github, 2024-12-14:
Did you read the readme ? It is wrote in bold that what you do is not correct

emrindustries, 2024-12-27:
Thanks Mate

I was reading the wrong README file.  My issue was the version of the IDE and after reading the right file I had no issues at all.
thanks heaps for your help

closed 2024-12-27 by emrindustries